realsenseai/librealsense #4586: DKMS failure 2.25

kevindehecker, 2019-08-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   |  | 
| Operating System & Version |   Linux Ubuntu 18  | 
| Kernel Version (Linux Only)    |  5.0.0-23-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25                      | 
| Language                            |                       | 
| Segment			|                      | 

### Issue Description
The installation of 2.25 fails on ubuntu 18.04 5.0.0-23-generic:

```
sudo apt install librealsense2*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'librealsense2-dbg' for glob 'librealsense2*'
Note, selecting 'librealsense2-dev' for glob 'librealsense2*'
Note, selecting 'librealsense2' for glob 'librealsense2*'
Note, selecting 'librealsense2-gl-dbg' for glob 'librealsense2*'
Note, selecting 'librealsense2-gl-dev' for glob 'librealsense2*'
Note, selecting 'librealsense2-gl' for glob 'librealsense2*'
Note, selecting 'librealsense2-utils' for glob 'librealsense2*'
Note, selecting 'librealsense2-dkms' for glob 'librealsense2*'
Note, selecting 'librealsense2-udev-rules' for glob 'librealsense2*'
The following NEW packages will be installed:
  librealsense2 librealsense2-dbg librealsense2-dev librealsense2-dkms librealsense2-gl librealsense2-gl-dbg librealsense2-gl-dev
  librealsense2-udev-rules librealsense2-utils
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
Need to get 4122 kB/114 MB of archives.
After this operation, 172 MB of additional disk space will be used.
Get:1 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 librealsense2-dkms all 1.3.6-0ubuntu0 [4122 kB]
Fetched 4122 kB in 4s (1128 kB/s)             
Selecting previously unselected package librealsense2:amd64.
(Reading database ... 282600 files and directories currently installed.)
Preparing to unpack .../0-librealsense2_2.25.0-0~realsense0.1332_amd64.deb ...
Unpacking librealsense2:amd64 (2.25.0-0~realsense0.1332) ...
Selecting previously unselected package librealsense2-dbg:amd64.
Preparing to unpack .../1-librealsense2-dbg_2.25.0-0~realsense0.1332_amd64.deb ...
Unpacking librealsense2-dbg:amd64 (2.25.0-0~realsense0.1332) ...
Selecting previously unselected package librealsense2-dev:amd64.
Preparing to unpack .../2-librealsense2-dev_2.25.0-0~realsense0.1332_amd64.deb ...
Unpacking librealsense2-dev:amd64 (2.25.0-0~realsense0.1332) ...
Selecting previously unselected package librealsense2-dkms.
Preparing to unpack .../3-librealsense2-dkms_1.3.6-0ubuntu0_all.deb ...
Unpacking librealsense2-dkms (1.3.6-0ubuntu0) ...
Selecting previously unselected package librealsense2-gl:amd64.
Preparing to unpack .../4-librealsense2-gl_2.25.0-0~realsense0.1332_amd64.deb ...
Unpacking librealsense2-gl:amd64 (2.25.0-0~realsense0.1332) ...
Selecting previously unselected package librealsense2-gl-dbg:amd64.
Preparing to unpack .../5-librealsense2-gl-dbg_2.25.0-0~realsense0.1332_amd64.deb ...
Unpacking librealsense2-gl-dbg:amd64 (2.25.0-0~realsense0.1332) ...
Selecting previously unselected package librealsense2-gl-dev:amd64.
Preparing to unpack .../6-librealsense2-gl-dev_2.25.0-0~realsense0.1332_amd64.deb ...
Unpacking librealsense2-gl-dev:amd64 (2.25.0-0~realsense0.1332) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../7-librealsense2-udev-rules_2.25.0-0~realsense0.1332_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.25.0-0~realsense0.1332) ...
Selecting previously unselected package librealsense2-utils:amd64.
Preparing to unpack .../8-librealsense2-utils_2.25.0-0~realsense0.1332_amd64.deb ...
Unpacking librealsense2-utils:amd64 (2.25.0-0~realsense0.1332) ...
Setting up librealsense2:amd64 (2.25.0-0~realsense0.1332) ...
Setting up librealsense2-udev-rules:amd64 (2.25.0-0~realsense0.1332) ...
Setting up librealsense2-dkms (1.3.6-0ubuntu0) ...
Loading new librealsense2-dkms-1.3.6 DKMS files...
Building for 5.0.0-23-generic
Building initial module for 5.0.0-23-generic
Error! Bad return status for module build on kernel: 5.0.0-23-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.6/build/make.log for more information.

Loading the modified modules into kernel...modprobe: FATAL: Module hid_sensor_accel_3d is in use.
Current status:
librealsense2-dkms, 1.3.6: added
nvidia, 390.116, 4.15.0-55-generic, x86_64: installed
nvidia, 390.116, 4.18.0-25-generic, x86_64: installed
nvidia, 390.116, 5.0.0-23-generic, x86_64: installed
mod: videodev 			
mod: uvcvideo 			 version: 1.1.1
mod: hid_sensor_gyro_3d 	
mod: hid_sensor_accel_3d 	
Setting up librealsense2-dev:amd64 (2.25.0-0~realsense0.1332) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Setting up librealsense2-dbg:amd64 (2.25.0-0~realsense0.1332) ...
Setting up librealsense2-gl:amd64 (2.25.0-0~realsense0.1332) ...
Setting up librealsense2-gl-dev:amd64 (2.25.0-0~realsense0.1332) ...
Setting up librealsense2-gl-dbg:amd64 (2.25.0-0~realsense0.1332) ...
Setting up librealsense2-utils:amd64 (2.25.0-0~realsense0.1332) ...
Processing triggers for libc-bin (2.27-3ubuntu1) ...
```

The contents of make.log:

```
DKMS make.log for librealsense2-dkms-1.3.6 for kernel 5.0.0-23-generic (x86_64)
di  6 aug 2019 15:28:51 CEST
make: Entering directory '/usr/src/linux-headers-5.0.0-23-generic'
arch/x86/Makefile:146: CONFIG_X86_X32 enabled but no binutils support
scripts/Makefile.build:42: /var/lib/dkms/librealsense2-dkms/1.3.6/build/5.0.0/drivers/media/usb/uvc/Makefile: No such file or directory
make[1]: *** No rule to make target '/var/lib/dkms/librealsense2-dkms/1.3.6/build/5.0.0/drivers/media/usb/uvc/Makefile'.  Stop.
Makefile:1778: recipe for target 'uvcvideo.ko' failed
make: *** [uvcvideo.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-5.0.0-23-generic'
```

ev-mp, 2019-08-06:
@kevindehecker hello, DKMS currently provides for LTS versions [ up to v4.18](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages). 
Adding support for v5.0 is scheduled in future releases.

kevindehecker, 2019-08-06:
Right, I didn't even notice my kernel got upgraded. This was an automatic update...

kevindehecker, 2019-08-06:
So actually this is not related to the 2.25 release at all. I'll change the title. I have 2.25 working fine now with the downgraded 4.18 kernel.

mjscosta, 2019-08-08:
We've also encountered this issue. Just to double check, we are not using DKMS modules, just the librealsense packages to access the camera.

The camera can no longer be accessed via librealsense2 in kernel version 5?

kevindehecker, 2019-08-08:
For me the camera itself works fine (I get images etc), but the advanced features don't work. Like the hardware clock.

mimfgg, 2019-08-09:
Hi! When would this update be scheduled for?

fnoop, 2019-08-09:
Hi, this is a pretty major issue as anyone who runs the latest LTS and keeps it up to date will now be broken!

milapila, 2019-08-14:
@kevindehecker After downgrading to 4.18, did you remove the 5.0-kernel? I ran into exactly the same problem and changed the kernel I'm booting from by changing the GRUB_DEFAULT.  
`uname -r` now returns 4.18.0-25-generic, but the error remains when entering
 `sudo apt-get install librealsense2-dkms`. Any suggestions?

kevindehecker, 2019-08-14:
Nope, I didn't uninstall 5.0 kernel.

s-hemer, 2019-08-22:
Adding to @fnoop's comment: With the latest, clean 18.04 LTS one is now completely lost, cmp #4669 (beside manual downgrading the kernel of course).

kevindehecker, 2019-10-01:
I just updated everything, now running kernel 5.0.0-29-generic with realsense api v 2.29, and this problems seems to be fixed!

drzraf, 2019-10-07:
Installation for Ubuntu via package (inc. dkms) is still failing.
Shouldn't this bug be reopened, or at least redirect to another one?

stefan-enway, 2019-10-07:
which kernel/realsense version are you using? did you treid the -29/ api v 2.29 version as kevin did?

drzraf, 2019-10-07:
Even better : `5.0.0-31`

```
$ cat /var/lib/dkms/librealsense2-dkms/1.3.6/build/make.log
DKMS make.log for librealsense2-dkms-1.3.6 for kernel 5.0.0-31-generic (x86_64)
Mon 07 Oct 2019 06:10:45 PM -03
make: Entering directory '/usr/src/linux-headers-5.0.0-31-generic'
arch/x86/Makefile:146: CONFIG_X86_X32 enabled but no binutils support
scripts/Makefile.build:42: /var/lib/dkms/librealsense2-dkms/1.3.6/build/5.0.0/drivers/media/usb/uvc/Makefile: No such file or directory
make[1]: *** No rule to make target '/var/lib/dkms/librealsense2-dkms/1.3.6/build/5.0.0/drivers/media/usb/uvc/Makefile'.  Stop.
make: *** [Makefile:1780: uvcvideo.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-5.0.0-31-generic'
``

stefan-enway, 2019-10-08:
thats the point. check the 2.29 dpkms package and look which kernel version it pachtes. as long as intel does only provide this kind of closely tied patches, you cant just take the latest.

stefan-enway, 2019-10-08:
anotherway would be to do a sourcecode patch and use checkinstall https://wiki.ubuntuusers.de/checkinstall/ to create a debian package from that.

kevindehecker, 2019-10-08:
Yeah, indeed. I just tried with another system and with -31 the problem is there again. I'm starting to doubt if it was ever fixed, as the system I tested this with had somewhat messed up config to start with...

I have reopened the issue, to at the moment I lag the time to keep track of everything.

ev-mp, 2019-10-16:
@dapi hello,
DKMS package [currently supports LTS kernels up to 4.18](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages) for Ubuntu 18. Support for v5.0 is yet to be provided.
Currently you need install the kernel patches for v5.0 with the [developer's script](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) - 
```
./scripts/patch-realsense-ubuntu-lts.sh
```

daenny, 2019-12-04:
@ev-mp is there any update on the kernel v5.0 support? Ubuntu LTS 18.04.3 is shipping with v5.0 as default kernel, so it would be nice to have the dkms working for that release.

ev-mp, 2020-02-06:
DKMS package with kernels 5.0 and 5.3 support is available for download.
FYI

MoBaT, 2020-02-20:
> DKMS package with kernels 5.0 and 5.3 support is available for download.
> FYI

Hey ev-mp,

I used the ./scripts/patch-realsense-ubuntu-lts.sh to build the kernal modules for kenel 5.3.0-40-generic and I'm having alot of trouble getting data from certain sensors. Whenever I try to use the "get_frame_metadata()" call, I get "UVC header is not available". I also cannot get data from the GYRO or ACCEL sensors. In the realsense viewer, I see:

>  20/02 13:32:13,409 WARNING [139862447482624] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 20/02 13:32:13,409 WARNING [139862455875328] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 20/02 13:32:18,414 WARNING [139862447482624] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 20/02 13:32:18,414 WARNING [139862455875328] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 20/02 13:32:23,416 WARNING [139862455875328] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 20/02 13:32:23,416 WARNING [139862447482624] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 20/02 13:32:34,036 WARNING [139862816564992] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 20/02 13:32:34,036 WARNING [139862824957696] (backend-hid.cpp:598) iio_hid_sensor: Frames didn't arrived within 5 seconds
 20/02 13:32:43,213 WARNING [139862803977984] (sensor.cpp:333) Frame received with streaming inactive,Depth0, Arrived,1582230763105.654053 1582230763111.562012
 20/02 13:34:36,205 WARNING [139862439089920] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
 20/02 13:34:47,977 WARNING [139862439089920] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.

I've been having trouble getting the realsense camera to work completely on my Ubuntu 18.04.04 machine.

kevindehecker, 2020-02-25:
@ev-mp I can confirm that the DKMS package installs without errors with kernel 5.3.0-40-generic. 
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2-dkms librealsense2-udev-rules
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/10,2 MB of archives.
After this operation, 54,6 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 381916 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.13-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.13-0ubuntu1) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.32.1-0~realsense0.1976_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.32.1-0~realsense0.1976) ...
Setting up librealsense2-udev-rules:amd64 (2.32.1-0~realsense0.1976) ...
Setting up librealsense2-dkms (1.3.13-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.13 DKMS files...
Building for 5.3.0-40-generic
Building initial module for 5.3.0-40-generic
Secure Boot not enabled on this system 25/02 12:36:36,898 WARNING [140498295576320] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
Done.

uvcvideo:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

videodev.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

hid-sensor-accel-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

hid-sensor-gyro-3d.ko:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

depmod...

DKMS: install completed.

Loading the modified modules into kernel... complete
Current status:
librealsense2-dkms, 1.3.13, 5.3.0-40-generic, x86_64: installed
nvidia, 390.116, 4.15.0-88-generic, x86_64: installed
nvidia, 390.116, 4.18.0-25-generic, x86_64: installed
nvidia, 390.116, 5.3.0-28-generic, x86_64: installed
nvidia, 390.116, 5.3.0-40-generic, x86_64: installed
mod: videodev 			 version: realsense2-dkms
mod: uvcvideo 			 version: 1.1.2.realsense-1.3.13
mod: hid_sensor_gyro_3d 	 version: realsense2-dkms
mod: hid_sensor_accel_3d 	 version: realsense2-dkms
```

However, the advanced metadata is not available out of the box after this, the realsense viewer gives me: ` 25/02 12:36:36,898 WARNING [140498295576320] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details`
Among others. I guess I'll open a new issue for this.

@MoBaT this issue is about the dkms package giving errors... Suggest you open another issue as well.

closed 2020-02-25 by kevindehecker

mikepurvis, 2020-03-06:
I'm still having the same error message from the original post, with `1.3.11-0ubuntu1` and kernel version `5.0.0-41.45`. Any thoughts?