realsenseai/librealsense #4589: Setting depth exposure manually takes too long

zomboir, 2019-08-06:
Hi there,
I am having issues when manually setting the depth exposure values of D435/D435i cameras. 

Irrespective of the value that I'm setting the rs2::sensor::set_option(...) call itself takes somewhere between 40-60ms. Assuming 30fps operation roughly two frames are skipped while setting this parameter. Any ideas why is this so slow, or how could I make this adjustment faster? I need to get as close to 30fps as possible. I tested different USB ports, different cables, using an externally powered USB hub, but it all the same. 

On the other hand adjusting the color exposure works fine, the rs2::sensor::set_option(...) call takes between 3-4ms as I would expect. 

Here is the code that I used:
```c++
    //created earlier
    //std::shared_ptr<rs2::sensor> m_depth_sensor;

    auto start = std::chrono::high_resolution_clock::now();
    m_depth_sensor->set_option(RS2_OPTION_EXPOSURE, 100);
    auto end = std::chrono::high_resolution_clock::now();

    std::chrono::duration<double, std::milli> totalElapsed = end - start;
    std::cout << "Setting depth exposure took " << totalElapsed.count() << "\n";
```
Output:
Setting depth exposure took 57.8007
Setting depth exposure took 40.8539
Setting depth exposure took 65.1323
Setting depth exposure took 55.6867
Setting depth exposure took 52.2004
Setting depth exposure took 58.9615
Setting depth exposure took 60.1332
Setting depth exposure took 55.9723
Setting depth exposure took 56.7763
Setting depth exposure took 58.1829

Is this the correct way to use the manual exposure control? Do I need to set any other parameter when I'm using manual exposure control? The RGB sensor has this "Auto Exposure Priority" for example...

Cheers,
Gergely

------------------------------------------------------------
 Camera Model                       D435 / D435i 
 Firmware Version                   5.11.6.250
 Operating System & Version  Win10
 Platform                                 PC
 SDK Version                            2.19.1
 Language                               C/C++

MartyG-RealSense, 2019-08-06:
If you are setting the exposure only once at the start of the stream, you could perhaps set your custom value before the stream starts by loading in near the start of your script, before the pipeline start instruction, a 'json' custom preset configuration file that you have created. 

A json file can be created by setting values in the RealSense Viewer and then exporting those settings in a json that can be loaded into the Viewer or into your own program.

[image]

Here is an example json for the 'Default' preset for the D435 camera model.

https://raw.githubusercontent.com/wiki/IntelRealSense/librealsense/d400_presets/DefaultPreset_D435.json

The link below gives example code for loading in the json.

https://github.com/IntelRealSense/librealsense/issues/3277

zomboir, 2019-08-07:
I aim to control the exposure value on the fly, while the camera is streaming. I have a custom made "auto-exposure" method which worked very well for my use-case.

At the moment I am adjusting the depth&color gain and exposure values right after a frame was obtained. So far the color settings and the depth gain adjustment works without any frame loss. As soon as I try to set the depth exposure too, the frame rate drops from 30 to 20.

MartyG-RealSense, 2019-08-07:
There is a case here of frame rate changing when the exposure is changed:

https://github.com/IntelRealSense/librealsense/issues/1957

zomboir, 2019-08-07:
Hi @MartyG-RealSense
I've seen #1957 but that topic is about controlling the RGB sensor's exposure value. There are some good questions raised though.

To test whether my issue is related to the USB3 throughput I dropped the frame sizes to 640x480 and disabled all other realtime adjustments. Now I am only setting the depth exposure value right after each frame was captured. 

I tested a few different fps configurations, but the framerate always drops if I enable depth exposure modification.

640x480@90fps
- frame-rate drops from 90 to 64
- rs2::sensor::set_option(...) call takes on average 29ms.

640x480@60fps
- frame-rate drops from 60 to 41
- rs2::sensor::set_option(...) call takes on average 35ms.

640x480@30fps
- frame-rate drops from 30 to 22
- rs2::sensor::set_option(...) call takes on average 66ms.

640x480@15fps
- frame-rate drops from 15 to ~12
- rs2::sensor::set_option(...) call takes either
    - ~1.5ms
    - ~67ms
    - ~133ms
- sometimes I get an exception with error code 0x8007001f  "A device attached to the system is not functioning"

Is there a certain point in the image acquisition pipeline where the depth exposure can be changed? It looks like rs2::sensor::set_option(...) is waiting on something, and the time it takes is related to the framerate somehow.

MartyG-RealSense, 2019-08-07:
I thought carefully about your problem but I don't have any further ideas.  One of the Intel guys on this forum can hopefully provide you with helpful guidance.   Good luck!

dorodnic, 2019-08-20:
Hi @zomboir 
Error 0x8007001f is most likely bug on our part, but otherwise, I think this is somewhat the expected behavior. I'm not an expert in how this is implemented in firmware, but changing exposure from host can take several frames. 
The real question is what is the problem you are trying to solve? If you are interested in some kind of HDR workflow, there is hardware capability to cycle between several fixed values of firmware on frame by frame basis.

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

Hi @zomboir, do you have any further question on this topic?

zomboir, 2019-09-11:
Hi there, 
sorry about the late reply. I have a use-case where I need to make a moving target correctly exposed. The visible size and position of the target in the FOV can change during capture. That's why I need the custom exposure control method which I developed using OpenCV. 

So far I've found that if I change the exposure every 6th frame I can get around 25fps reliably. Of course, it would be better not to have so much delay and to run the capture near 30fps, but 25fps is good enough for my application.

If you say the framedrop is expected when using manual exposure control - and the delay cannot be reduced - then we can leave it at that and close the ticket.  

The best would be to have a function to request a change of exposure (non-blocking, with no immediate effect), and let the firmware decide when is the right time to apply the new value with the smallest possible latency.

RealSenseCustomerSupport, 2019-09-11:
----------------------------------------------

Hi @zomboir,
Thank you so much for the update and I am really glad to hear you have found a solution that works for your application.

I will go ahead and close this ticket now.

closed 2019-09-12 by RealSenseCustomerSupport