realsenseai/librealsense #4590: calibrating t265 and d435 together

svabhishek29, 2019-08-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { t265, d435 } | 
| Firmware Version                   |  | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC |
| Language                            |  {python }                          | 
| Segment			|  {AR }                          | 

hi, 

i was checking out the mount designed by intel to use d435 and t265 together, i have a custom mount, how can i calibrate the setup myself so i can use them together?

Thanks

schmidtp1, 2019-08-06:
Hi, you can try https://github.com/IntelRealSense/librealsense/pull/4355.

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

Hi @svabhishek29 - Were you able to take schmidtp1 advice on trying the PR?

svabhishek29, 2019-09-08:
yes, it worked well...the suggestion was able to help me solve the problem.
Thanks

RealSenseCustomerSupport, 2019-09-13:
----------------------------------------------

Glad to hear that it worked for you.

closed 2019-09-13 by RealSenseCustomerSupport