realsenseai/librealsense #4591: Path generated by rs-data-collect (csv) does not match path generated by rs-record (bag)

peterwatsonnm, 2019-08-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | t265 | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.4.38                                         | 
| Platform                                 | Jetson tx2  |
| SDK Version                            |  2.0                          | 
| Language                            |  shell                         | 
| Segment			|  Robot                       | 

### Issue Description
Hi all,

I've been trying to gather data in a .csv format using the rs-data-collect tool.  

Here is my code calling the function:
`rs-data-collect -c /home/nvidia/intelrealsense/librealsense/tools/data-collect/config.csv -f /home/nvidia/Experiments/t265_data/Test_$filename.csv -t 15 `

And here is my config file, config.csv
[image]

When I plot the 3DOF_X and 3DOF_Y values in Excel in an x y plot, I get something like this:
[image]

Whereas the path shown by the bag file in realsense-viewer, with the camera following the same physical track, looks like this:
[image]

Same with comparing it to a gps:

T265 .csv data:
[image]

GPS data following the same physical track:
[image]

I've tried changing all of the physical variables in the testing setup, so I think this is a software issue or an issue with my code.  I have a feeling it is related to the config file because I don't really understand how that works.

I'm not a software guy, so go easy on me if this is a simple fix! I enjoy learning about this technology and I'm very grateful to all the helpful people on here!

ev-mp, 2019-08-06:
@peterwatsonnm , It seem you meant to record T265 positions and compare the results with a reference sensor. Instead, the `rs-data-collect` was configured to collect the inertial (acceleration and gyro) data which cannot be used to reproduce the path.
If that's the case then the proper way to preset the configuration file for the task is to put the following line
>GYRO,1,1,200,MOTION_XYZ32F

From the tool's [documentation](https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect#config-file-format)

peterwatsonnm, 2019-08-06:
@ev-mp thanks for the response!  Are you saying that rs-data-collect will not work at all for this task?  Or will it work with this line in the .config file? 

And, if it will not work at all, is there an alternative way to get the xy position in csv format? Thanks!

peterwatsonnm, 2019-08-06:
Using the line you posted in my config file has fixed the problem.  I can now make a graph of the position!  Thanks again, I will close the issue.

closed 2019-08-06 by peterwatsonnm