realsenseai/librealsense #4592: Can we change frequency of T265 from 200hz?

peterwatsonnm, 2019-08-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | Jetson tx2  |
| SDK Version                            |                           | 
| Language                            |                          | 
| Segment			|  Robot                        | 

### Issue Description

I am curious if we can change the frequency of the T 265 from 200 HZ.  This frequency is much to high for my project, and I'd like to lower it if possible, specifically for the rs-data-collect function to record data into the .csv file slower than 200 frames per second.

Thank you!

JvanZijl, 2019-08-06:
I don't mean to be rude, but since you have the T265 working on the Jetson tx2, could you maybe help me out in #4578? That would be great and thank you!

ev-mp, 2019-08-07:
@peterwatsonnm , the Pose output data rate is fixed and cannot be changed.
Note that the affect of streaming and parsing T265 positions is negligible in terms of CPU utilization even on low-end boards (i.e. `rs-data-collect` ), so you can apply an application filter to read only every N-th frame or so.
In case you want to minimize the resulted CSV then just modify `rs-data-collect` with this logic. Here is the place where it can be applied - https://github.com/IntelRealSense/librealsense/blob/master/tools/data-collect/rs-data-collect.cpp#L341-L343

peterwatsonnm, 2019-08-07:
@ev-mp thanks for the response.  I will try these solutions, but my main issue is the amount of data that the t-265 is sending through the cable.  I was hoping that a usb 2 could handle the pose data stream, but when streaming for 5+ minutes, the cable gets overloaded.  An upgrade to a usb 3 is in my future, but right now I only have a usb 2 cable that is the length I need.  Thanks for the info, though!

peterwatsonnm, 2019-08-07:
@JvanZijl commented :)

ev-mp, 2019-08-08:
@peterwatsonnm , can you elaborate what did you mean by
>I was hoping that a usb 2 could handle the pose data stream, but when streaming for 5+ minutes, the cable gets overloaded

Here are some numbers: the T265 frame size is 131 bytes, and 200fps results in 131x200=26,200 bytes/sec throughput. For reference a webcam streaming VGA at 30 fps produces 30*640*480*2(yuyv) = 3,686,400 bytes/sec (~3.5MB/sec). 
Is it possible that 0.7% of an entry-level webcam bandwidth would stress the TX2 and its USB controller, which even in USB2 mode designed for 480MBit=60 MBytes/sec?

peterwatsonnm, 2019-08-09:
When I run the camera through USB 2.0 cables, after streaming pose data with rs-data-collect for 2.5 minutes, the data in the CSV file is just displayed as #NONE or nan... Problem dissappears when usb 3 is used.  I have just been using usb 3 as a solution for now.  Thanks for the info, that is cool how low latency it is!

harville, 2019-08-22:
@peterwatsonnm Have you seen any NaN pose output since starting to use USB 3 cables?

closed 2019-09-18 by RealSenseCustomerSupport

neilyoung, 2019-10-23:
@harville I have.

harville, 2019-10-30:
@peterwatsonnm  If you need further support on occurrence of NaN pose output, please refer to issues [4518](https://github.com/IntelRealSense/librealsense/issues/4518) and [5101](https://github.com/IntelRealSense/librealsense/issues/5101), where an ongoing discussion continues as we work to resolve this.

jingshaojing, 2021-01-08:
my interest is how can i know the frequency of t265 pose data is 200hz? though i have known the imu frequency is 200hz,i want to find the offical explanition.

jingshaojing, 2021-01-08:
> my interest is how can i know the frequency of t265 pose data is 200hz? though i have known the imu frequency is 200hz,i want to find the offical explanition.

I have found the reference,but  I have the new question, when there is no new frame,the pose data is keeping as the last frame's data  or change to 0?