realsenseai/librealsense #4593: Corrupted Maps Being Exported

JBBee, 2019-08-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.6100 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC |
| SDK Version                            |  2.25.0                           | 
| Language                            | unity                         | 
| Segment			|  AR                         | 

### Issue Description
I'm seeing an issue with 2.25.0 where it's possible to be unable to import a map that was previously exported by the T265.  

To reproduce this in the viewer all I have to do is start and map an area, then stop and export that map.  Then import the same file I exported and see if it was successful.  If it was successful I run the device for a little with the imported map until it relocalizes.  Then stop and repeat the steps from the beginning.  After roughly five times the map import fails with the message below.  

Invalid Value in rs2_import_localization_map(sensor:0000022AA6265BE0, lmap_blob:R, blob_size:849021):
import localization failed, map size 849021

Thanks.

JBBee, 2019-08-07:
I tested it a little further, and when the map comes out corrupted the T265 is actually in a bad state where you can't interact with it further until you unplug it and plug it back in.

RealSenseCustomerSupport, 2019-09-11:
----------------------------------------------

Thanks @JBBee, after a while I was able to reproduce the issue as well

RealSenseCustomerSupport, 2019-11-12:
----------------------------------------------

Hi @JBBee - are you still running into corrupted maps issue?

JBBee, 2019-11-12:
Hi @RealSenseCustomerSupport actually we are still seeing corrupted maps being exported every so often in 2.30.0.  Please see below.

Steps to reproduce in the viewer:
- Start with a freshly plugged in T265.  Start the module and walk around the room for a bit, for maybe 20 seconds.  Stop the module and export the localization map to a file.
- Import the file just exported and run it for a short while, again about 20 seconds.  It should relocalize at some point.  Stop the module
- Start the module and walk around for 20 seconds again.  Stop the module and export the localization map.  
- Import the localization map, and here I'm seeing it fail to import, the map is corrupt.

Thanks.

radfordi, 2019-11-12:
Thanks for the procedure @JBBee.  While we try and duplicate this, could you provide the final map that fails to import?

JBBee, 2019-11-12:
Sure @radfordi , here is the corrupted map.  Thanks again.

[map.zip](https://github.com/IntelRealSense/librealsense/files/3838150/map.zip)

neilyoung, 2019-11-12:
Interesting. Will try to reproduce tomorrow.

radfordi, 2019-11-12:
Thanks @JBBee.  I'm able to duplicate the issue with that map.

radfordi, 2019-11-13:
@JBBee, #5234 should fix this.  It also doubles the internal map size as a bonus.

neilyoung, 2019-11-13:
@radfordi I can confirm the issue. I additionally noticed no relocation info on the third run, step 3. Import failed too in step 4. Currently cloning your repo. While running Cmake I got this trace:

`--Info: REALSENSE_VERSION_STRING=2.17.0`

Not sure if it is just cosmetic.

Anyway. After build I can't make your version run. Starting realsense-viewer just gives this 

```
decades@ubuntu:~/Documents/radfordi-librealsense/build/tools/realsense-viewer$ ./realsense-viewer
 13/11 11:46:12,797 ERROR [140529635838528] (backend-v4l2.cpp:518) Cannot access /sys/class/video4linux
 13/11 11:46:12,804 ERROR [140529635838528] (backend-v4l2.cpp:518) Cannot access /sys/class/video4linux
 13/11 11:46:13,068 ERROR [140529635838528] (backend-v4l2.cpp:518) Cannot access /sys/class/video4linux
 13/11 11:46:13,285 ERROR [140529635838528] (backend-v4l2.cpp:518) Cannot access /sys/class/video4linux
```

Since your PR is not merged yet, I have to wait for it I guess. Not a problem.

JBBee, 2019-11-13:
Thanks @radfordi I'll grab it once it's merged.

radfordi, 2019-11-13:
@neilyoung, those message aren't related to T265.   When running `cmake` you should have seen:
```txt
-- Info: REALSENSE_VERSION_STRING=2.30.0
```
so I'd guess you don't have my branch checked out.  Try doing `git checkout t265-release` and then building again.

neilyoung, 2019-11-14:
True. Works. No corrupted map anymore. Double size confirmed. I thought I would have filed an issue, but I can't actually find it: At least under Ubuntu there is no .raw extension on save added. So the file open dialog doesn't find it. It is just an annoyance, just so you know.

It's getting better day by day :) Great PR.

JBBee, 2019-12-03:
@radfordi 0.2.0.879 in 2.31.0 works great.  I'm not seeing any corrupted maps and the relocalization is faster.  Thanks!

closed 2019-12-03 by JBBee

radfordi, 2019-12-03:
Thanks for following up @JBBee.