realsenseai/librealsense #4594: Pose-apriltag T265 example not working in different platforms

AnthonyKyung11, 2019-08-06:
Specifications | Version / Model
------------ | -------------
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.6100 | 
| Operating System & Version  | Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    | 4.15.0-50-generic  | 
| Platform                                  | Intel NUC 5i7RYH  |
| SDK Version                            |  librealsense  2.25.0   | 
| Language                                |  C++ | 
 

### Issue Description

Hi I'm running the librealsense example pose-apriltag in the Intel NUC with the previous specifications and it works fine, but if I run the same code in the Samsung Artik 530 with the following specifications it does not work, it gives me the following error: **Frame didn't arrived within 15000**.

The code enables both streams, the pose and the fisheye as it show in the following code: 
                  cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
                   // Enable both image streams
                  cfg.enable_stream(RS2_STREAM_FISHEYE, 1, RS2_FORMAT_Y8);
                  cfg.enable_stream(RS2_STREAM_FISHEYE, 2, RS2_FORMAT_Y8); 

and it gives me the previous error, but if I only enable one stream pose or both fisheye images, it does not gives me any problem.

I tried:
    - enabling the pose stream after certain number of image frames
    - enabling the fisheye images after certain number of pose frames
    - enabling and disabling image stream after certain number of image frames 
    - enabling and disabling pose stream after certain number of pose frames
    - enabling and disabling image stream after certain number of pose frames
    - enabling and disabling pose stream after certain number of image frames

and all of the previous tries give me the same error: **Frame didn't arrived within 15000**.

Do you have any idea what could be causing this?

Specifications | Version / Model
------------ | -------------
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.6100 | 
| Operating System & Version  | Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    | 4.4.71-0530GC0F-44U-01Q1  | 
| Platform                                  | Artik 530  |
| SDK Version                            |  librealsense  2.25.0   | 
| Language                                |  C++ |

hoangthien94, 2019-08-09:
Hi, just my 2 cents: does the new platform have USB3.0 and if yes, are you using it to interface with the T265? For image streams, you would need USB3.0, as described [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#what-are-the-system-requirements).

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

Hi @AnthonyKyung11

Were you able to verify the USB3 connection on the Samsung platform you outline?

Thanks

AnthonyKyung11, 2019-09-06:
Yes, that was the issue, we were trying to see the image using USB 2.0. Thank you very much.

closed 2019-09-06 by AnthonyKyung11