realsenseai/librealsense #4600: External Camera Trigger input with Pixhawk

Ken-Thorburn, 2019-08-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |                            | 
| Platform                                 | Intel Nuc  |
| SDK Version                            |  2.17| 
| Language                            |  {N/A }                          | 
| Segment			|  Robot                         | 

### Issue Description

Hello, 

I am trying to trigger the rgb camera of the Intel Realsense D415 with a Pixhawk's camera trigger plugin in order to correctly time stamp images so as to provide accurate data to a VIO algorithm. The Pixhawk's signal and ground are connected to the VSYNC and ground port of the D415's external connector. When in the Realsense-viewer I have set Inter-cam-sync mode to 2 and the Auto Exposure Priority off, but the camera continues to trigger  frames on its own and not recognize the Pixhawk's signals. 

Any help would be appreciated

[image]

oafolabi, 2019-08-15:
I'm facing a similar issue too, anyhelp is appreciated.

Ken-Thorburn, 2019-08-16:
We haven't figured out a way to enable it just yet. That window is in QGroundControl, an app to configure and fly a PX4 based autopilot, under settings, camera trigger.

RealSenseCustomerSupport, 2019-09-04:
----------------------------------------------

Hi

Please note that, the RealSense D400 provides camera "Sync" mode, not "Single Trigger" mode.
When you configure camera to "mode 2", it is set as a slave camera. And user is supposed to provide an external pulsing source - but must be **continuous** pulsing (i.e. 30Hz). So the slave camera can be sync'd to the pulsing. The pulsing frequency must be equal or 1/N (N=2, 3, 4....) to camera native frame rate (i.e. 30FPS). For example, when camera is set at 30FPS, the external pulsing frequency can be 30Hz, 15Hz, 7.5Hz .....

The slave camera will be running freely upon its own internal clock if there is no external pulsing.

Thanks

closed 2019-09-19 by RealSenseCustomerSupport