realsenseai/librealsense #4601: Android: calculate pointcloud from depthmap?

gene-alex, 2019-08-07:
Camera Model | D415
Firmware Version | 05.11.01.100
Operating System & Version | Android 9
Platform | Samsung 8
SDK Version | 2.0
Language | Java

First issue and not sure if this is the appropriate forum.  Are there plans to extend the functionality of the Android wrapper to include the pointcloud calculation from the depthmap?  Thanks.

dorodnic, 2019-08-12:
Hi @gene-alex 
Welcome to librealsense community :)

> Are there plans to extend the functionality of the Android wrapper to include the pointcloud calculation from the depthmap

I think it is already available - [Pointcloud.java](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/librealsense/src/main/java/com/intel/realsense/librealsense/Pointcloud.java)
Just not added yet to the App / examples

erikgallegos, 2019-08-12:
Hi @dorodnic

From your link to the `Pointcloud` class, it looks like you can only create the point cloud object, is it also possible to actually generate the points?

I've been looking for something like `Pointcloud.calculate()` but it's not available.
 
Any help would be appreciated, thanks!

erikgallegos, 2019-08-12:
@dorodnic  I've since figured out that `Pointcloud.process(frame)` is what I was looking for and I have the following (using Kotlin):

> val byteArray = ByteArray(depthFrame.width * depthFrame.height)
>
> val frame = Pointcloud(StreamType.DEPTH).process(depthFrame)
>
> frame.getData(byteArray)

It's my understanding that the point cloud data is now stored in `byteArray` so I went through each byte in `byteArray` and plotted the points:

[image]

As you can see this doesn't look right, I'm unsure what the correct way is to process/interpret the data and there is not much documentation to look at. Could you shed some light on the proper way to do this?

matkatz, 2019-08-16:
Hi @erikgallegos 
The byte array need to be converted to float array in order to get the pointcloud.
I'm planning to add an API thatdoes that for you soon.
You can check this branch [android-pointcloud-support](https://github.com/matkatz/librealsense/tree/android-pointcloud-support) which already has this API, see [GLPointsFrame](https://github.com/matkatz/librealsense/blob/android-pointcloud-support/wrappers/android/librealsense/src/main/java/com/intel/realsense/librealsense/GLPointsFrame.java) (`points.getVertices()` line 67) for usage example.

dorodnic, 2019-08-27:
Hope this can be closed, feel free to open new issue if you have any follow-up questions

closed 2019-08-27 by dorodnic