realsenseai/librealsense #4603: USB SCP overflow, no frames received

jungladicitta, 2019-08-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.15.0-55-generic                                  | 
| Platform                                 | PC  |
| SDK Version                            |  2.25.0                       | 
| Language                            |  ---                        | 
| Segment			|  Robot                         | 

I use two cameras d435 and t265.
I run realsense-viewer and sometimes I have problem with d435. I see that no frames received and red messages on top-right.
[image]

Sometimes, when it works totally fine, i see notification that "Incomplete video frame detected"
[image]

d00m3dd00d, 2019-08-09:
I often run into this as well with my D415. A few tips that seems to work around it for now:

-Adjusting settings while streaming seems to cause this most often. Be sure to turn off a feed before adjusting settings.
-Using a setting that is not supported prevents the stream from loaded, both on Android and Windows.
-Unplugging and plugging the device back in will reboot it's internal software, unfreezing it.
-If all else fails, reset defaults from the settings gear in Windows from the top right.

jungladicitta, 2019-08-09:
@d00m3dd00d 
yeah, only unplugging and plugging does help from time to time

d00m3dd00d, 2019-08-09:
I did find that my USB-C to USB-C cable was working as USB 2.1 and the
cable included with the camera was doing the same with my PC due to
something else plugged in on the bus, I'm thinking that is likely the cause
of my problems. Perhaps make sure you're using a 3.1 gen 2 cable, that may
also sole it. The error can be related to bandwidth/stream errors.

jungladicitta, 2019-08-14:
i have ordered new cables. 1 meter and 1.5 meter. next week

d00m3dd00d, 2019-08-14:
Can confirm that new cable with 3.1 gen 2 with 10mb/s bandwidth support resolved all problems for me, software on Android now functions as expected. Only one application (that is out of date in relation to SDK updates) had any issues.

RealSenseCustomerSupport, 2019-09-04:
----------------------------------------------

@jungladicitta, is this resolved with the new cable? If so, can we close it?

jungladicitta, 2019-09-05:
@RealSenseCustomerSupport 

yes

closed 2019-09-05 by jungladicitta