realsenseai/librealsense #4604: pipe start throw exception,and rs2_get_librealsense_exception_type return 720

barcelona-delpy, 2019-08-08:
My camera info is here:
[image]

I use vs2015 in win10,and librealsense SDK is 2.0
### Issue Description
code is like this:

rs2::pipeline *temp_pipe = new rs2::pipeline(ctx);
rs2::config temp_cfg;
temp_cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16, 30);
temp_cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
temp_cfg.enable_device(temp_sn);
temp_pipe->start(temp_cfg);  //this line throw a exception and rs2_get_librealsense_exception_type  return 720

when comment all temp_cfg.enable_stream line,temp_pipe->start(temp_cfg)  success,why?
What should i do to fix this problem?

ev-mp, 2019-08-08:
@barcelona-delpy hello, 
What is the actual camera type that you're using - is it D430 or D435(i) ?  Can you attach a screenshot ?
In the above screenshot the camera is recognized as D430, a model w/o RGB sensor. If that's correct then the following line in the code is not applicable
>~~temp_cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);~~

You can list the supported streaming resolutions and formats with `rs-enumerate-devices -m`

RealSenseCustomerSupport, 2019-09-11:
----------------------------------------------

Hi @barcelona-delpy,

Please provide us with following information about the configuration you have experienced issues with:

  RealSense SDK 2.0 build version

  Camera type (D430/ D435/ D435i/etc.)

Thank you,
David

ev-mp, 2019-09-19:
@barcelona-delpy hello, do you have the requested data for analysis? 
Please update

closed 2019-09-23 by RealSenseCustomerSupport