realsenseai/librealsense #4606: Multiple Cameras with pyrealsense2 TypeError: get_info(): incompatible function arguments.

LLUWE19, 2019-08-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)  |                               | 
| Platform                                 | PC |
| SDK Version                           |  { legacy / 2.2.25 }                          | 
| Language                               |  python                      | 
| Segment			        |  others                           | 

Hi, 

I am trying to automate the opening of multiple devices. I have managed to open multiple devices following this code:
https://github.com/IntelRealSense/librealsense/issues/1735

However I want to read and load cameras automatically. I have tried following the C++ tutorial and swapping for python but cannot extract the serial number. e.g.

```
# Create librealsense context for managing devices
ctx = rs.context()

pipelines = []

for d in ctx.query_devices():
    print(d)
    pipe = rs.pipeline(ctx)
    cfg = rs.config()
    cfg.enable_device(d.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER))
    pipe.start(cfg)
    pipelines.append(pipe)

```

```
// Start a streaming pipe per each connected device
for (auto&& dev : ctx.query_devices())
{
    rs2::pipeline pipe(ctx);
    rs2::config cfg;
    cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
    pipe.start(cfg);
    pipelines.emplace_back(pipe);
}
```

from C++ tutorial
> To map the specific device to the newly-allocated pipeline we define rs2::config object, and assign it with the device's serial number.

from config documentation:
> Select a specific device explicitly by its serial number, to be used by the pipeline.

From this information I guess I have to manually enter serial numbers?

What should I enter in the RS2_CAMERA_INFO_SERIAL_NUMBER?
Is there a way to extract the serial number automatically? Or load devices automatically?

TypeError: get_info(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.device, info: pyrealsense2.camera_info) -> str

Hope question makes sense and thanks for your time

closed 2019-08-09 by LLUWE19

BryanVision, 2019-08-14:
Did you get this solved?

lramati, 2019-08-20:
@BryanVision RS2_CAMERA_INFO_* is how these enum values are accessed in C++. In Python, you need rs.camera_info.* (see docs [here](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.camera_info.html#pyrealsense2.camera_info))