realsenseai/librealsense #4607: patching issue for 4.19.0 kernel

manukharshan, 2019-08-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { / D400 } | 
| Firmware Version                   |  | 
| Operating System & Version |   4.19.0-xilinx-v2019.1  | 
| Kernel Version (Linux Only)    |  4.19.0                                       | 
| Platform                                 | xilinx zynq zcu104   |
| SDK Version                            | librealsense2                          | 
| Language                            |  C                         | 
| Segment			|                            | 

### Issue Description
After following the installation instructions found at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Manually i run the patch for the UVC kernel module.
patch -p1 < /home/test/librealsense/scripts/realsense-camera-formats.patch

I'm getting the following error 

Output:
patch -p1 < /home/test/librealsense/scripts/realsense-camera-formats.patch
patching file drivers/media/usb/uvc/Makefile
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- saving rejects to file drivers/media/usb/uvc/Makefile.rej
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 289 with fuzz 2 (offset 141 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 174 with fuzz 2 (offset 55 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 621.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej

Please give your suggestions for this issue

Thanks and Regards,
Manu

ev-mp, 2019-09-02:
@manukharshan , unfortunately we maintain kernel patches for Ubuntu LTS versions only which do not include 4.19.
Kernel patches are not a prerequisite for running the SDK and in many cases can be skipped.
When building from sources that you can invoke "-DFORCE_LIBUVC=true" flag that provides user-space implementation for most kernel patches.
See #4484 for details

RealSenseCustomerSupport, 2019-09-10:
----------------------------------------------

Hi @manukharshan,

Do you need any further clarifications regarding this topic?

Thank you!

closed 2019-09-13 by manukharshan