realsenseai/librealsense #4608: D435 Frame Rate Drops Above Certain Resolutions Running on Arm64

rosslannen, 2019-08-08:
| Required Info | |
| --- | --- |
| Camera Model | D435 (depth module only) | 
| Firmware Version                   | 5.09.02.00 | 
| Operating System & Version  | Yocto Linux | 
| Kernel Version (Linux Only)   | 4.9 | 
| Platform                                  | Arm 64 NXP IMX 8m quad core |
| SDK Version                           |  2.23.0  | 
| Language                                | C++ | 

### Issue Description
When running the camera at resolutions above 640x480 at both 15 and 30 frames per second, the frame rate of the incoming data drops dramatically, with long and inconsistent times between frames. I have included the code that was used to reproduce this bug, the parameters it was run with, and the compiled results of those tests, along with the raw output of the program attached as a file.

This is only reproducible on the Arm 64 system. All resolutions and frame rate configurations run as expected on an Intel x64 platform running Ubuntu with the same code piece. The Arm system is attached to the camera over a usb 3.2 port.

Because the problem shows itself so dramatically with a relatively small resolution change, and is not affected by frame rate, it does not indicate it is a bandwidth/processing power constraint.

### Code
CameraTest.cpp:

```c++
#include <chrono>
#include <iostream>
#include <sstream>
#include <tuple>
 
#include <librealsense2/rs.hpp>
 
std::tuple<int, int, int, int> parse_args(int argc, char **argv) {
  if (argc < 5) {
    std::cout << "Usage: LuciSensors WIDTH HEIGHT RATE ITERATIONS" << std::endl;
    exit(1);
  }
 
  auto arg_stream = std::stringstream();
  arg_stream << argv[1] << ' ' << argv[2] << ' ' << argv[3] << ' ' << argv[4];
 
  int width, height, rate, iterations;
  arg_stream >> width >> height >> rate >> iterations;
 
  return std::make_tuple(width, height, rate, iterations);
}
 
int main(int argc, char **argv) {
  auto [width, height, rate, iterations] = parse_args(argc, argv);
 
  std::cout << "Values:" << '\n'
            << "    Width: " << width << '\n'
            << "    Height: " << height << '\n'
            << "    Rate: " << rate << '\n'
            << "    Iterations: " << iterations << std::endl;
 
  auto context = rs2::context();
 
  const auto &&device = context.query_devices().front();
 
  auto config = rs2::config();
  config.enable_device(device.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
  config.enable_stream(RS2_STREAM_DEPTH, width, height, RS2_FORMAT_Z16, rate);
 
  auto pipeline = rs2::pipeline();
  pipeline.start(config);
 
  std::cout << "Starting frame reads" << std::endl;
  auto start_time = std::chrono::system_clock::now();
  auto last_frame_time = start_time;
  for (int i = 0; i < iterations; ++i) {
    auto _frameset = pipeline.wait_for_frames();
 
    // print the time since the last frame arrived
    auto now = std::chrono::system_clock::now();
    auto elapsed = std::chrono::duration_cast<std::chrono::milliseconds>(
        now - last_frame_time);
    std::cout << "Time since last frame: " << elapsed.count() << std::endl;
    last_frame_time = now;
  }
  auto end_time = std::chrono::system_clock::now();
  std::cout << "Stopping frame reads" << std::endl;
 
  auto run_time = std::chrono::duration_cast<std::chrono::milliseconds>(
      end_time - start_time);
 
  auto average_frame_rate = 1000.0 * iterations / run_time.count();
 
  std::cout << "Average frame rate: " << average_frame_rate << std::endl;
 
  return 0;
}
```

### Results
All results are run with 30 iterations.

| Frame Width (pixels) | Frame Height (pixels) | Configured Frame Rate (Hz) | Actual Frame Rate (Hz) | Pass/Fail |
| --- | --- | --- | --- | --- |
| 480 | 270 | 15 | 13.363 | Pass |
| 480 | 270 | 30 | 26.115 | Pass |
| 640 | 360 | 15 | 13.345 | Pass |
| 640 | 360 | 30 | 24.116 | Pass |
| 640 | 480 | 15 | 13.333 | Pass |
| 640 | 480 | 30 | 26.087 | Pass |
| 848 | 480 | 15 | 1.396 | FAIL |
| 848 | 480 | 30 | 6.356 | FAIL |
| 1280 | 720 | 15 | TIMEOUT | FAIL |
| 1280 | 720 | 30 | TIMEOUT | FAIL |

### Raw Program Output
[camera-data.txt](https://github.com/IntelRealSense/librealsense/files/3483234/camera-data.txt)

dorodnic, 2019-08-09:
Hi @rosslannen 
There are certainly opportunities for better optimization on all platforms, but I’d start by checking standard HD/FHD Usb webcam with Linux standard tools like yavta.
Not sure why you rule out bandwidth / processing power constraints so quickly. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

kevswims, 2019-08-09:
Hi @dorodnic 

We ruled out bandwidth / processing power because the 848x480 at 15 FPS which runs at less than 2Hz requires less bandwidth than the 640x480 at 30 FPS which runs fine.

The issues seem correlated very well with the change in resolution but not frame rate.

Regards,

Kevin

rosslannen, 2019-08-15:
Hello @dorodnic!

As @kevswims said, bandwidth/processing power issues seem very unlikely in this scenario, given the large variance between frame rates with such small changes.

For the 640 x 480 @ 15 fps, that is a total of 4,608,000 pixels per second that are being processed. In comparison, 848 x 480 @ 15 fps is a total of 6,105,000 pixels per second. That is an increase of under 50% total data size, but results in a rate that is 1/10th as fast as it should be. This is further shown by the fact that 640 x 480 @ 30 fps, at 9,216,000 pixels per second, runs perfectly fine, which is at a higher total throughput than the higher resolutions at slower speeds. If performance struggled at 640 x 480 @ 30 fps, then hardware load would be more reasonable guess as to what is happening.

These numbers indicate to me that it is a bug somewhere in the code, and not a hardware performance issue, that causes the higher resolutions to slow down to a crawl. Unless there is a portion of the code that scales at O(n^5) (or worse) with respect to the number of pixels per frame, then the numbers above cannot be explained with performance alone.

Another thing indicating it is not bandwidth, I was able to copy a 300 MB file over the usb on our hardware in 1.56 seconds, indicating our usb can handle at least 1.5 Gb per second, which should be more than enough to handle depth data from a D430 at any of the supported resolutions and frame rate. I'm guessing the usb port can handle more, but the speed was bottlenecked by the reads off of the flash drive I was copying off of.

Thanks for your help!

Ross

rosslannen, 2019-08-19:
Hi @GUY,

Thank you for the suggestions for debugging using `yavta`. Our system was able to pull frames from a third party 720p webcam without any issues.

I was also able to pull data directly from a D435 camera using `yavta`, and not using the librealsense APIs. I was able to run the camera at 848x480 resolution at both 15 and 30 frames per second, with no noticeable slowdown. Shown is an attached output from running at 848x480 at 30fps. This does not mirror the behavior seen using librealsense with the earlier provided code sample.

Is there anything else you would like me to try out to help isolate the problem?

``` bash
# yavta /dev/video8 --capture --size=848x480 --time-per-frame=1/30
Device /dev/video8 opened.
Device `Intel(R) RealSense(TM) Depth Ca' on `usb-xhci-hcd.1.auto-1.3' is a video output (without mplanes) device.
Video format set: Z16  (2036315a) 848x480 (stride 1696) field none buffer size 814080
Video format: Z16  (2036315a) 848x480 (stride 1696) field none buffer size 814080
Current frame rate: 1/90
Setting frame rate to: 1/30
Frame rate set: 1/30
8 buffers requested.
length: 814080 offset: 0 timestamp type/source: mono/SoE
Buffer 0/0 mapped at address 0xffff7f781000.
length: 814080 offset: 815104 timestamp type/source: mono/SoE
Buffer 1/0 mapped at address 0xffff7f6ba000.
length: 814080 offset: 1630208 timestamp type/source: mono/SoE
Buffer 2/0 mapped at address 0xffff7f5f3000.
length: 814080 offset: 2445312 timestamp type/source: mono/SoE
Buffer 3/0 mapped at address 0xffff7f52c000.
length: 814080 offset: 3260416 timestamp type/source: mono/SoE
Buffer 4/0 mapped at address 0xffff7f465000.
length: 814080 offset: 4075520 timestamp type/source: mono/SoE
Buffer 5/0 mapped at address 0xffff7f39e000.
length: 814080 offset: 4890624 timestamp type/source: mono/SoE
Buffer 6/0 mapped at address 0xffff7f2d7000.
length: 814080 offset: 5705728 timestamp type/source: mono/SoE
Buffer 7/0 mapped at address 0xffff7f210000.
0 (0) [-] none 0 814080 B 253391.864609 253391.871792 8.148 fps ts mono/SoE
1 (1) [-] none 1 814080 B 253391.897987 253391.905118 29.960 fps ts mono/SoE
2 (2) [-] none 2 814080 B 253391.931319 253391.938437 30.001 fps ts mono/SoE
3 (3) [-] none 3 814080 B 253391.964675 253391.971807 29.980 fps ts mono/SoE
4 (4) [-] none 4 814080 B 253391.998015 253392.005055 29.994 fps ts mono/SoE
5 (5) [-] none 5 814080 B 253392.031371 253392.038407 29.980 fps ts mono/SoE
6 (6) [-] none 6 814080 B 253392.064719 253392.071729 29.987 fps ts mono/SoE
7 (7) [-] none 7 814080 B 253392.098138 253392.105146 29.923 fps ts mono/SoE
8 (0) [-] none 8 814080 B 253392.131431 253392.138479 30.036 fps ts mono/SoE
9 (1) [-] none 9 814080 B 253392.164786 253392.171823 29.981 fps ts mono/SoE
10 (2) [-] none 10 814080 B 253392.198124 253392.205171 29.996 fps ts mono/SoE
11 (3) [-] none 11 814080 B 253392.231525 253392.238570 29.939 fps ts mono/SoE
12 (4) [-] none 12 814080 B 253392.264863 253392.271904 29.996 fps ts mono/SoE
13 (5) [-] none 13 814080 B 253392.298209 253392.305216 29.989 fps ts mono/SoE
14 (6) [-] none 14 814080 B 253392.331532 253392.338581 30.009 fps ts mono/SoE
```

Thank you again for all your help!

dorodnic, 2019-08-20:
Hi @rosslannen 
Thank you for the feedback. It's good to know yavta is able to stream Z16 without problems.
To understand what is causing problems with librealsense application, could you please:

a. Enable UVC logs
```
Enable:
sudo echo 0xffff > /sys/module/uvcvideo/parameters/trace

Disable:
sudo echo 0x0 > /sys/module/uvcvideo/parameters/trace
```
Output will be dumped to dmesg. 

b. Enable librealsense logs in Debug severity. Simplest way is to add `rs2::log_to_file("filename", RS2_LOG_SEVERITY_DEBUG)`.

These two steps will log frames at V4L2 - librealsense boundary. 
My thinking is to check that Z16 and only Z16 is being requested by librealsense, and see we get any partial frames warnings from the driver.

rosslannen, 2019-08-20:
Hi @dorodnic ,

I have attached the dmesg uvc  output, and the realsense logs. Let me know what you see, or if there is anything in particular I should be looking for in them.

[dmesg-uvc-dump.txt](https://github.com/IntelRealSense/librealsense/files/3521875/dmesg-uvc-dump.txt)
[realsense-log.txt](https://github.com/IntelRealSense/librealsense/files/3521879/realsense-log.txt)

RealSenseCustomerSupport, 2020-02-28:
----------------------------------------------

Hi Jered, Ross, David and Kevin,

Hope you are doing well! It's been a while since we've spoken.How your project is going?
We still have one open ticket "D435 Frame Rate Drops Above Certain Resolutions Running on Arm64" and we'd like to know if it can be closed until the time you have any updates. When you do have updates/issues, we can re-engage this discussion.

Thank you,
David

kevswims, 2020-02-28:
Hi David,

We actually just started an internal investigation into this issue. I have been able to use a USB analyzer to capture what is happening when we have framerate issues. I will attach a PDF that goes into more detail about the experiments I have conducted so far.

Summary:
It seems that the realsense camera is not sending complete frames up. On the incomplete frames the camera sends back a bulk transaction packet with zero data length which signals to the host to end the transaction for that frame. We have been able to replicate this on NXP's IMX8M devkit which does not contain a hub so it that rules out our specific hardware or any interaction with a hub inline. 

I can send you the raw analyzer data separately as it is too large to attach on Github.

Thanks,

Kevin

[PD-RealsenseIncompleteVideoFrameDetected-280220-2211.pdf](https://github.com/IntelRealSense/librealsense/files/4269329/PD-RealsenseIncompleteVideoFrameDetected-280220-2211.pdf)

closed 2020-03-03 by rosslannen

kruftindustries, 2020-03-30:
Did someone ever figure out why the camera is sending incomplete frames? I have been making modifications to v4l2 and recompiling kernel to try and resolve this assuming it was a software issue to no avail before finding this. 

This issue was closed and there is no one assigned,  however I'm not sure it is resolved.

kruftindustries, 2020-04-01:
I believe I have found the cause of this issue. After attempting to change which arm core processes uvcvideo irq interrupts I make a mistake and searching some of the error messages make gave when compiling the kernel lead me to find some documentation on the subject. 

Short version:
most arm processors aren't powerful enough to handle the kind of throughput these cameras have with a single core. The solution is to add USB packets to a queue and process on multiple cpu cores.

Solution:
switch to a newer kernel version

Long version:
Search for linux kernel commit 6f0e5fd39143a59c22d60e7befc4f33f22aeed2f
[https://lore.kernel.org/patchwork/cover/871285/](https://lore.kernel.org/patchwork/cover/871285/)

kevswims, 2020-04-01:
@kruftindustries have you tried this and has it fixed the issue for you? I don't see those patches you reference on either the 4.14 or 4.19 kernels that NXP uses for our board. They are present on the 5.4 branch but that's not something I can easily test out.

Additionally, if this was just a throughput issue wouldn't we expect to see the issue at low resolutions and higher frame rates and not at higher resolutions with lower framerates?

That commit you posted 6f0e5fd39143a59c22d60e7befc4f33f22aeed2f is the parent of the patches but the patches do not seem to have been accepted until much later on.

kruftindustries, 2020-04-01:
Yes I have tried with 5.4.28 and can confirm realsense-viewer works almost
as good as on a regular desktop. No warning messages about >2mSec until
depth color and IR enabled at resolutions like 1280x720x60, native
resolution with yuv is rock solid.

There are some differences in the development workflow and initramfs takes
~30s to unzip during reboot. I had assumed I broke something the first few
times until I noticed this.

After reviewing torvalds/linux tags in GitHub it looks like they made some
major changes to uvcvideo in some of the release candidates preceding
kernel 4.20. There will be some performance improvements in that release
but can't quite tell if they are enough. Might be worth a try if it's
possible

kevswims, 2020-04-07:
Hi @kruftindustries 

I ended up taking a 4.19 kernel and cherry-picking the patchset you linked to onto that kernel and we saw some improvements but it is still not working correctly.

The patches seemed to help the problem at some resolutions for the depth data but not others for us. We still have some weird behavior that is not bandwidth dependent where at the 848x480 resolution there are some dropped frames but it now runs, at 640x360 (yes, a lower resolution) it is way worse and drops over half the frames, and I have yet to see an error at 640x480. I have tested these at both 30 and 15 fps and the framerate makes no difference in behavior.

There still seems to be a bug with some of the depth resolutions.

Thanks,

Kevin