realsenseai/librealsense #4609: Segfault when trying to declare colorizer before starting the pipeline

DJ73, 2019-08-09:
I am using D435i and trying to create a class that handles all RealSense API calls.
In that, I tried to declare processing blocks like colorizer, decimation, etc so as to call them for each frame. When I perform `pipe.start()` after this and then try to apply the filters, I get a segfault.

Is this intended behaviour? Can I get around it?

PS it works if I declare it after `pipe.start()` but the my code has a single function to get a frame and process it which has to declare the filters for every frame.

dorodnic, 2019-08-09:
Hi @DJ73 
This is certainly not expected behavior.
Feel free to share parts of your code.
Also, sdk examples may be helpful for reference 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

DJ73, 2019-08-09:
Sorry, must have been something with my code. I tried making a small test program (attached below) and that works. But I can't figure out how to avoid declaring align every time I get a frameset. I tried adding it as a class member but that gave me the error `no matching function for call to ‘rs2::align::align()’`

```
#include <librealsense2/rs.hpp>
#include <opencv2/opencv.hpp>
#include <cv-helpers.hpp>
#include <iostream>
#include <string>

typedef struct sConfigs
{
  rs2_stream stream_type;
  int stream_index;
  int width;
  int height;
  rs2_format format;
  int framerate;

  sConfigs()
  {
    stream_type   = RS2_STREAM_ANY;
    stream_index  = -1;
    width         = 0;
    height        = 0;
    format        = RS2_FORMAT_ANY;
    framerate     = 0;
  }
} stream_configs;

class TestWrapper
{
  private:
    stream_configs colorcfg;
    stream_configs depthcfg;

    rs2::config m_cfg;
    rs2::pipeline m_pipe;
    rs2::pipeline_profile m_pipeProfile;
    rs2::colorizer m_colorizer;
    rs2::decimation_filter m_decimation;
    rs2::spatial_filter m_spatial;
    rs2::temporal_filter m_temporal;

  public:
    TestWrapper(){}
    
    int setconfig(std::string prop, std::string value)
    {
           if (prop == "COLOR_WIDTH")
      {
        colorcfg.width = std::stoi(value);
      }
      else if (prop == "COLOR_HEIGHT")
      {
        colorcfg.height = std::stoi(value);
      }
      else if (prop == "COLOR_FORMAT")
      {
        if (value == "RGB8")
        {
          colorcfg.format = RS2_FORMAT_RGB8;
        }
        else if (value == "BGR8")
        {
          colorcfg.format = RS2_FORMAT_BGR8;
        }
      }
      else if (prop == "COLOR_FPS")
      {
        colorcfg.framerate = std::stoi(value);
      }
      else if (prop == "DEPTH_WIDTH")
      {
        depthcfg.width = std::stoi(value);
      }
      else if (prop == "DEPTH_HEIGHT")
      {
        depthcfg.height = std::stoi(value);
      }
      else if (prop == "DEPTH_FORMAT")
      {
        if (prop == "Z16")
        {
          depthcfg.format = RS2_FORMAT_Z16;
        }
      }
      else if (prop == "DEPTH_FPS")
      {
        depthcfg.framerate = std::stoi(value);
      }
      return 0;
    }

    int startPipe()
    {
      m_cfg.enable_stream(RS2_STREAM_COLOR, -1,
                          colorcfg.width ,
                          colorcfg.height,
                          colorcfg.format,
                          colorcfg.framerate);
      m_cfg.enable_stream(RS2_STREAM_DEPTH, -1,
                          depthcfg.width ,
                          depthcfg.height,
                          depthcfg.format,
                          depthcfg.framerate);
      m_pipeProfile = m_pipe.start(m_cfg);

      m_pipe.wait_for_frames();
      return 0;
    }

    int getFrames()
    {
      auto frames = m_pipe.wait_for_frames();
      rs2::align m_align(RS2_STREAM_COLOR);
      frames = m_align.process(frames);
      frames = m_decimation.process(frames);
      frames = m_spatial.process(frames);
      frames = m_temporal.process(frames);

      auto color_image = frame_to_mat(frames.get_color_frame());

      auto depth_frame = frames.get_depth_frame();
      auto colorized_image = frame_to_mat(m_colorizer.process(depth_frame));

      auto depth_image = depth_frame_to_meters(m_pipe, depth_frame);
      depth_image = depth_image * 1000;
      depth_image.convertTo(depth_image, CV_16U);

      cv::imshow("Color", color_image);
      cv::imshow("Depth", depth_image);
      cv::imshow("Colorized", colorized_image);
      cv::waitKey(1);

      return 0;
    }

    int stopPipe()
    {
      m_pipe.stop();
    }

};

int main()
{
  TestWrapper dummy;

  dummy.setconfig("COLOR_WIDTH", "960");
  dummy.setconfig("COLOR_HEIGHT", "540");
  dummy.setconfig("COLOR_FORMAT", "BGR8");
  dummy.setconfig("COLOR_FRAMERATE", "60");
  dummy.setconfig("DEPTH_WIDTH", "848");
  dummy.setconfig("DEPTH_HEIGHT", "480");
  dummy.setconfig("DEPTH_FORMAT", "Z16");
  dummy.setconfig("DEPTH_FRAMERATE", "90");

  dummy.startPipe();
  for (int i = 0; i < 120; i++)
  {
    dummy.getFrames();
  }
  dummy.stopPipe();

  return 0;
}
```

closed 2019-08-09 by DJ73

dorodnic, 2019-08-09:
No worries, we are happy to assist 🙂
I think in your class you need to add something like: 
```
TestWrapper::TestWrapper() 
     : m_colorizer(), m_align(RS_...)
{
}
```
With C++ style constructor initialization.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

DJ73, 2019-08-12:
That worked. Thank you