realsenseai/librealsense #4610: Accessing Camera Information Python

LLUWE19, 2019-08-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |  Win (8.1/10) | 
| Platform                                 | PC |
| SDK Version                            |  2.23.0                          | 
| Language                            | python                          | 
| Segment			|  others                        | 

### Issue Description

I am trying to figure out how to retrieve the serial number from the camera and have been trying to the get_info() but end up with this error.

>TypeError: get_info(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.device, info: pyrealsense2.camera_info) -> str

Any help would be greatly appreciated

MartyG-RealSense, 2019-08-09:
Have you seen the similar approach to yours in the link below please that a D415 user created for Pyrealsense2?  They use **rs.camera_info.serial_number** and r**s.camera_info.name** to get the serial number details.

https://stackoverflow.com/questions/50724909/pyrealsense2-librealsense-sdk-2-0-choose-cam-from-serial-number

LLUWE19, 2019-08-09:
I hadnt seen it, was exactly what I was looking for. Guess i forgot to use 'rs' 

Thanks MartyG very grateful

closed 2019-08-09 by LLUWE19