realsenseai/librealsense #4611: Cannot change any controls for RGB Camera on MacOS

pzoltowski, 2019-08-09:
| | |
|---|---|
|**librealsense**|2.25.0 RELEASE|
|**OS**|Mac OS|
|**Name**|Intel RealSense D435I|
|**Serial Number**|843112070584|
|**Firmware Version**|05.11.06.250|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|852323051288|

MacOS 10.14.4 

I tried to change RGB Camera controls via realsense-viewer app but its seems it doesn't work. GUI sliders doesn't have any knob and when I click e.g. backlight compensation / auto exposure priority or gamma it throws:

```
UNKNOWN in rs2_set_option(options:0x7fbc67450330, option:Backlight Compensation, value:6):
libusb_control_transfer(...) returned LIBUSB_ERROR_PIPE

UNKNOWN in rs2_set_option(options:0x7fbc67450330, option:Auto Exposure Priority, value:6):
libusb_control_transfer(...) returned LIBUSB_ERROR_PIPE

UNKNOWN in rs2_set_option(options:0x7fbc67450330, option:Gamma, value:6):
libusb_control_transfer(...) returned LIBUSB_ERROR_PIPE
```

When I try to change exposure it sets to 6. When I try to input parameters manually it throws:
`60 is out of bounds [6, 6]`
(for all settings it restricts bounds to [6, 6])

For IR Camera stream most settings have knob and works except 'GAIN' - it throws:
```
UNKNOWN in rs2_set_option(options:0x7fbc67450040, option:Gain, value:6):
libusb_control_transfer(...) returned LIBUSB_ERROR_PIPE
```
I tried also modify 'python-rs400-advanced-mode-example.py' to change settings but changing anything doesn't seems to affect anything (would be good to update this sample to not just modify settings but also open any video stream to clarify how to adjust settings via python)

I mostly require to track fast motions that's why I need to adjust settings manually. I also couldn't find information for the following information:
1) what's the range and units for 'Exposure'? Is it camera shutter time in microseconds?
2) what's the range and units for 'Gain'? Is it related to camera ISO?
3) how does 'Auto Exposure Priority' work?
4) what's difference between the following settings?:
    ```
"controls-autoexposure-auto": "True",
    "controls-color-autoexposure-auto": "True",

```
5) What's the focus distance of d435 and optimal focus field of depth?

pzoltowski, 2019-08-14:
does anyone can confirm have the same issue and manage to reproduce it?

dorodnic, 2019-08-14:
Yes, I can confirm the bug. Will roll out a fix as soon as possible 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

dorodnic, 2019-10-15:
This is being addressed as part of [#4975](https://github.com/IntelRealSense/librealsense/pull/4975), but its a large change and takes time to be integrated. Basic testing on few Macs look good.

pzoltowski, 2019-10-15:
That's a great news. Can this already be tested from current master branch or requires updated firmware?

pzoltowski, 2019-11-06:
I have tested again with latest 2.30.0 version and it seems to work now on my macbook with MacOS Mojave - both adjusting exposure for IR camera and adjusting all controls for RGB camera. Great!

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

Hi @pzoltowski,

Do you need any further assistance with this issue?

Thank you!

closed 2020-01-29 by RealSenseCustomerSupport