realsenseai/librealsense #4615: Writable Calibration Parameters for D400 devices

waiyc, 2019-08-10:
Hi,

Are we allow to modify camera's projection parameters and other parameters not stated in the document via RealSense customRW tool? As the  [Intel Calibration User Guide](https://dev.intelrealsense.com/docs/intel-realsensetm-d400-series-calibration-tools-user-guide) stated the writable calibration parameters are camera matrix and distortion (for intrinsic) translation and rotation ( for translation).

MartyG-RealSense, 2019-08-10:
The public version of the Dynamic Calibrator tool only calibrates extrinsics.  The 'OEM' version of the tool calibrates both extrinsics and intrinsics.

The OEM Calibrator is part of the $1500 OEM calibration system aimed at OEM manufacturing facilities and engineering departments and is available in Intel's online click store (the majority of the price is the cost of manufacturing the physical calibration board).  If you have an Intel Non Disclosure Agreement (NDA) then you may be able to obtain the software on its own by contacting your Intel representative through the special Zendesk support channel for NDA customers.  

If you are not an NDA customer and would like to become one, I would recommend posting a message on the Intel Support RealSense forum so that a support agent can get you started on the registration process.

https://support.intelrealsense.com/hc/en-us/community/topics

waiyc, 2019-08-10:
Hi @MartyG-RealSense 

Thank you for the reply. I wanted to build a custom calibration to calibrate intrinsic and extrinsic parameters. Therefore, I am concerning about writing the calibrated camera parameters into the device. Is there any camera parameters that is only accessible to NDA customer but not open to public ?

MartyG-RealSense, 2019-08-10:
As the OEM software is available as part of the $1500 calibration system from the public Click store, I would assume that the OEM-only intrinsic calibration functions are not an NDA secret and that the public version does not contain the intrinsic functions because the majority of users would not need to make use of them.    That is based on personal speculation though, as I have little knowledge about the exact workings of the OEM version.

For a deeper calibration system, you could investigate the software Vicalib, which has been used with the 400 Series cameras by Intel themselves.

https://github.com/arpg/vicalib

Both extrinsics and intrinsics were creatable and modifiable using the Python language from early on in the SDK, around February 2018.  There isn't much information about it though, other than the code in the SDK relating to this function:

https://github.com/IntelRealSense/librealsense/commit/658d672b97b64257afdbcf18cb60bcea38fe5f87#

dorodnic, 2019-08-19:
Hi @waiyc 

Key message regarding camera calibration - we do not want to encourage customers to mess with camera calibration since doing it wrong can easily diminish the quality of the product. 

That said - 

> Are we allowed to modify camera's projection parameters and other parameters not stated in the document via RealSense customRW tool?

You are certainly *allowed* to do so, it is your device. You should do this at your own risk (of making the camera unusable) and I'm not sure how this affects the warranty. (You could contact [support](https://www.intel.com/content/www/us/en/forms/support/intel-realsense-store-inquiry.html?store=irs) ahead of time to check if you will be able to return the camera if you mess up the calibration, I don't have the answer)

> Is there any camera parameters that is only accessible to NDA customer but not open to public ?

We encourage both public and NDA customers to use official calibration tools. As for the calibration format, it is public ([ds5-private.h:304-316](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-private.h#L304-L316)) but not super documented, due to the reasons above. I think left & right intrinsics and extrinsics + distortion coefficients is all there is to it, but let us know if you have any specific questions. 

If you do ever mess up, here is how you restore factory calibration - 
```cpp
rs2::debug_protocol dbg = rs2::context().query_devices().front();
std::vector<uint8_t> restore_def{
    0x14, 0, 0xAB, 0xCD, 0x61, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
};
auto res = dbg.send_and_receive_raw_data(restore_def);
```
Note that resetting to factory is not always the best solution since camera calibration changes over time. Resetting to factory and then running Dynamic Calibration or the upcoming On-Chip calibration should get you out of most troubles.

waiyc, 2019-08-21:
Hi @dorodnic , 

Thank you for the explanation. 
By the way, are you able to identify what might be the issues that causing [problem for converting raw YUY2 format to rgb/bgr](https://github.com/IntelRealSense/librealsense/issues/4646) in openCV?

dorodnic, 2019-08-27:
I didn't have an answer for that one of the top of my hat (it is more of a opencv question), but you can convert YUY to RGB using this SDK :)
See `rs2::yuy_decoder` class.

closed 2019-09-03 by dorodnic