realsenseai/librealsense #4617: Fail to build on Raspberry 4b

tom8zds, 2019-08-10:
--------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 } | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi/  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          |  
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi, I'm facing an error when compiling from the source code. The error came out from log.cpp and rs.cpp and option.cpp
Building CXX object CMakeFiles/realsense2.dir/src/log.cpp.o
cc1plus: warning: -Wformat-security ignored without -Wformat [-Wformat-security]
/home/pi/librealsense/src/log.cpp:14:12: error: ‘logger_type’ does not name a type; did you mean ‘res_type’?
     static logger_type<log_name> logger;
            ^~~~~~~~~~~
            res_type
/home/pi/librealsense/src/log.cpp: In function ‘void librealsense::log_to_console(rs2_log_severity)’:
/home/pi/librealsense/src/log.cpp:19:5: error: ‘logger’ was not declared in this scope
     logger.log_to_console(min_severity);
     ^~~~~~
/home/pi/librealsense/src/log.cpp:19:5: note: suggested alternative: ‘logf64’
     logger.log_to_console(min_severity);
     ^~~~~~
     logf64
/home/pi/librealsense/src/log.cpp: In function ‘void librealsense::log_to_file(rs2_log_severity, const char*)’:
/home/pi/librealsense/src/log.cpp:24:5: error: ‘logger’ was not declared in this scope
     logger.log_to_file(min_severity, file_path);
     ^~~~~~
/home/pi/librealsense/src/log.cpp:24:5: note: suggested alternative: ‘logf64’
     logger.log_to_file(min_severity, file_path);
     ^~~~~~
     logf64
make[2]: *** [CMakeFiles/realsense2.dir/build.make:1051: CMakeFiles/realsense2.dir/src/log.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/realsense2.dir/all] Error 2

filipematosinov, 2019-08-13:
Here is the solution https://github.com/IntelRealSense/librealsense/issues/4565

RealSenseCustomerSupport, 2019-09-03:
----------------------------------------------

Hi @tom8zds,

Is the issue resolved per the solution? If no additional support for this ticket, will close this.

tom8zds, 2019-09-03:
yeah, it works. Thank you.

closed 2019-09-04 by RealSenseCustomerSupport