realsenseai/librealsense #4621: Frames aligned in PointCloud Example?

pkr97, 2019-08-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)                                              | 
| Platform                                 | PC|
| SDK Version                            |  { 2.}                          | 
| Language                            |  {unity }                          | 

### Issue Description

I would like to get 3D position of an object in Color frame.
For that the steps i'm following are:
1. segment the object in Color Image.
2. get object position in color image (x,y) in pixel coordinate.
3. get 3d position for (x,y) from point cloud using index method.

Issue is that if the procedure will give accurate object position or not?
Are the depth frame and color frame are aligned automatically or we need to do it ourselves? if so then HOW? because the Align example is not intuitive and I'm not able to extract the concept.

If aligning is done automatically then no need to worry about manual aligning.

wodndb, 2019-08-16:
You can align color image and depth image using post processing.
Check the scene 'BackgroundSegmentationExample'.
RsProcessingPipe prefab has 'RsProcessingPipe' script and that has profile.
The profile has postprocessing block 'Rs Align'.

[image]

[image]

wodndb, 2019-08-16:
Yeah, I tested length measurement using D415 and it roughly works well.
You should use post processing block 'Rs Align' to align color frame to depth frame. there are several options and I think align to 'Depth' option is most accurate then other options.

pkr97, 2019-08-16:
Thanks for the reply. The day after I posted this, I was able to implement the very same thing of aligning to depth inside post-processing block. It works fine. 

Actually I'm using SR300. whose range is not far beyond 1.5 m .
Works well in that range.

closed 2019-08-16 by pkr97