realsenseai/librealsense #4623: I am getting an error while doing face detection with openCV DNN

shiva2410, 2019-08-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 |python.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
My code snippet

import numpy as np
import argparse
import cv2

print("[INFO] loading model...")
net = cv2.dnn.readNetFromCaffe('pk.prototxt.txt','sd.caffemodel')

image = cv2.imread('f7.jpg')

(h, w) = image.shape[:2]
blob = cv2.dnn.blobFromImage(cv2.resize(image, (300,300)), 1.0,
    (300,300), (104.0, 177.0, 123.0))
net.setInput(blob)
detections = net.forward()

for i in range(0, detections.shape[2]):
    # extract the confidence (i.e., probability) associated with the
    # prediction
    confidence = detections[0, 0, i, 2]

    if confidence > 0.5:
        box = detections[0, 0, i, 3:7] * np.array([w, h, w, h])
        (startX, startY, endX, endY) = box.astype("int")
        text = "{:.2f}%".format(confidence * 100)
        y = startY - 10 if startY - 10 > 10 else startY + 10
        cv2.rectangle(image, (startX, startY), (endX, endY),
            (0, 0, 255), 2)
        cv2.putText(image, text, (startX, y),
            cv2.FONT_HERSHEY_SIMPLEX, 0.45, (0, 0, 255), 2)
cv2.imshow("Output", image)
cv2.waitKey(0)

I am getting this 
error OpenCV(3.4.2) C:\Miniconda3\conda-bld\opencv-suite_1534379934306\work\modules\dnn\src\layers\convolution_layer.cpp:212: error: (-215:Assertion failed) blobs.size() != 0 in function 'cv::dnn::ConvolutionLayerImpl::getMemoryShapes'

Please help its very urgent and important

MartyG-RealSense, 2019-08-12:
There was a case on this forum with OpenCV DNN in 2028 that had a very similar error.  The link below contains the advice that was given by a member of the support team.

https://github.com/IntelRealSense/librealsense/issues/1604#issuecomment-390071039

jawad-tech, 2019-08-15:
I also faced this problem and i solved this problem by 

Downloading the .caffemodel and .prototxt file from MobileNet-SSD-master repository. The file name on this repository is different so you will first rename these files according to the files you already have in your build folder. Then replace the downloaded files in your build folder. Hope it helps

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

Hi @shiva2410,

Do you need any further clarifications regarding this topic?

shiva2410, 2019-09-25:
no

closed 2019-09-25 by RealSenseCustomerSupport