realsenseai/librealsense #4624: ./scripts/patch-realsense-ubuntu-lts.sh errors

yishanchuan, 2019-08-13:
Hi, guys. When I run `./scripts/patch-realsense-ubuntu-lts.sh`, I get following errors:
`/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.c:208:12: error: ‘UVC_GUID_FORMAT_L8’ undeclared here (not in a function); did you mean ‘UVC_GUID_FORMAT_Y8’?
   .guid  = UVC_GUID_FORMAT_L8,
            ^~~~~~~~~~~~~~~~~~
            UVC_GUID_FORMAT_Y8
/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.c:213:12: error: ‘UVC_GUID_FORMAT_D16’ undeclared here (not in a function); did you mean ‘UVC_GUID_FORMAT_Z16’?
   .guid  = UVC_GUID_FORMAT_D16,
            ^~~~~~~~~~~~~~~~~~~
            UVC_GUID_FORMAT_Z16
/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.c:218:12: error: ‘UVC_GUID_FORMAT_W10’ undeclared here (not in a function); did you mean ‘UVC_GUID_FORMAT_Y10’?
   .guid  = UVC_GUID_FORMAT_W10,
            ^~~~~~~~~~~~~~~~~~~
            UVC_GUID_FORMAT_Y10
/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.c:223:12: error: ‘UVC_GUID_FORMAT_CONFIDENCE_MAP’ undeclared here (not in a function); did you mean ‘V4L2_PIX_FMT_CONFIDENCE_MAP’?
   .guid  = UVC_GUID_FORMAT_CONFIDENCE_MAP,
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            V4L2_PIX_FMT_CONFIDENCE_MAP
/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.c:229:12: error: ‘UVC_GUID_FORMAT_RAW8’ undeclared here (not in a function); did you mean ‘UVC_GUID_FORMAT_L8’?
   .guid  = UVC_GUID_FORMAT_RAW8,
            ^~~~~~~~~~~~~~~~~~~~
            UVC_GUID_FORMAT_L8
/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.c:235:12: error: ‘UVC_GUID_FORMAT_RW16’ undeclared here (not in a function); did you mean ‘UVC_GUID_FORMAT_RG16’?
   .guid  = UVC_GUID_FORMAT_RW16,
            ^~~~~~~~~~~~~~~~~~~~
            UVC_GUID_FORMAT_RG16
/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.c:240:12: error: ‘UVC_GUID_FORMAT_BAYER16’ undeclared here (not in a function); did you mean ‘UVC_GUID_FORMAT_BG16’?
   .guid  = UVC_GUID_FORMAT_BAYER16,
            ^~~~~~~~~~~~~~~~~~~~~~~
            UVC_GUID_FORMAT_BG16
scripts/Makefile.build:330: recipe for target '/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.o' failed
make[1]: *** [/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc//uvc_driver.o] Error 1
make[1]: *** warting for tasks....
Makefile:1552: recipe for target '_module_/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc/' failed
make: *** [_module_/home/cliux/librealsense-master/ubuntu-bionic-master/drivers/media/usb/uvc/] Error 2
make: leave dir“/home/cliux/librealsense-master/ubuntu-bionic-master”
`
Ubuntu 18.04, kernel version 4.15

ev-mp, 2019-08-15:
@yishanchuan hello,
The Ubuntu LTS kernel trees have been shuffled recently with the introduction of kernel v5.0, for which the manual patch is yet to be fully adapter. ([WIP](https://github.com/ev-mp/librealsense/tree/bionic_update_hwe_5.0) branch).

But I recommend to switch to [DKMS](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) distribution package. This is the sustained solution for Ubuntu distro with LTS kernels. 
> sudo apt install librealsense2-dkms

In case you have to run the patches manually, (e.g. adding your-own kernel patches to media subsystem) then you can try the script from the following [PR](https://github.com/IntelRealSense/librealsense/pull/4509) by @themikem

ev-mp, 2019-09-02:
@yishanchuan , the PR [#4774](https://github.com/IntelRealSense/librealsense/pull/4774) that fixes the manual patching for Bionic is being evaluated.
Feel free to try and comment

ev-mp, 2019-09-19:
[Fixed](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#bug-fixes) in v2.28

closed 2019-09-19 by ev-mp