realsenseai/librealsense #4626: Realsense Viewer not working with Dell Inspiron 5559's embedded F200 Camera

tejashah88, 2019-08-13:
| Required Info                         |                                         |
|----------------------------------------|----------------------------------|
| Camera Model                       | F200                                | 
| Firmware Version                   | v2.25.0                            | 
| Operating System & Version  | Linux (Kubuntu 18.10)    | 
| Kernel Version (Linux Only)   |  4.18.0-25                        | 
| Platform                                 | PC                                    |
| SDK Version                          |  v2.25.0                            | 
| Language                               |  N/A                                 | 
| Segment			         |  N/A                                 | 

### Issue Description
I have a Dell Inspiron 5559 with an embedded F200 realsense camera. I was trying to get the realsense-viewer app working and it's not finding the camera. However, when I try a normal camera app, I can see the RGB and depth output normally.

MartyG-RealSense, 2019-08-13:
The RealSense SDK 2.0 and its RealSense Viewer program are not compatible with the F200 camera.  It supports the SR300 / Sr305, 400 Series and T265 cameras.

For the F200, the older 'Legacy' version of Librealsense or the '2016 R3' SDK for Windows should be used.

**Legacy Librealsense**
https://github.com/IntelRealSense/librealsense/tree/v1.12.1

**2016 R3**
https://software.intel.com/en-us/realsense-sdk-windows-eol

With the 'R3' SDK, you should download and install the core 'Essentials' module first, and then further optional modules to add additional features that you may need such as hand / face tracking.

tejashah88, 2019-08-13:
@MartyG-RealSense Ok thanks!

closed 2019-08-13 by tejashah88