realsenseai/librealsense #4627: Multi-cameras incomplete video frame

tonyhsiao1995, 2019-08-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.15.0-55-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.23.0                           | 
| Language                            |   --                           | 
| Segment			|  Robot                           | 

### Issue Description
I am trying to connect eight D415 cameras to the PC. I used PCI-E v3 (x4) USB 3.0 port to connect all cameras. The PCI-E v3 (x4) supports 3.94 GB/s bandwidth.  According to the table, one camera for 1280x720 resolution needs 885 Mbps. Theoretically, 8 cameras need 7,080 Mbps which is much smaller than 3.94 GB/s. However, when I connect more than six cameras, the Realsense-viewer will show "incomplete video frame" sometimes. Is my understanding wrong?
[image]

MartyG-RealSense, 2019-08-13:
Could you tell us please about the specification of the PC that you are attaching the cameras to?  If you are running 4 or more cameras on one PC then it should ideally have a spec of i7 processor or better in order to cope with it.

MartyG-RealSense, 2019-08-13:
Thank you very much for the information.  That is a very good spec.  Are you using the official short cables that came with the cameras or a cable of your own choice please?

tonyhsiao1995, 2019-08-13:
own choice. Official short USB + Extension USB cables

MartyG-RealSense, 2019-08-13:
Ok, thank you.  Using your own cable with RealSense cameras can sometimes be a lottery due to variations in quality.  If you are using a cable that is 2 m or more then you may experience performance reduction or no functioning at all unless it is a high quality cable.

Also, I recalled that the following advice had previously been given about incomplete frames by Dorodnic the RealSense SDK Manager:

https://github.com/IntelRealSense/librealsense/issues/1240#issuecomment-368760734

dorodnic, 2019-08-13:
> Realsense-viewer will show "incomplete video frame" sometimes

The question is how frequent is sometimes. USB Bulk mode does not guaranty quality-of-service from the OS, hence when underlying USB buffers get full (due to temporary peak in CPU utilization elsewhere in the system) some frames will get dropped. In our tests we check that number of frame drops does not go over 5%, but occasional single drops can happen even with one camera. 

In particular, if you write a program that consumes 100% CPU on all available cores, you will see degradation in performance of most USB devices.

tonyhsiao1995, 2019-08-13:
Thanks for the reply.

closed 2019-08-13 by tonyhsiao1995

wodndb, 2019-08-16:
@tonyhsiao1995 
Could you tell me what USB hub on your computer?
I'm finding best USB hub for D415 cameras.

MartyG-RealSense, 2019-08-16:
In Intel's paper on setting up sync between multiple cameras, they used an AmazonBasics mains powered hub for their tests and provided the link below.

https://www.amazon.com/dp/B00DQFGH80/ref=asc_df_B00DQFGH805315309

wodndb, 2019-08-16:
@MartyG-RealSense Thanks a lot

tonyhsiao1995, 2019-08-29:
@wodndb 
I am using something like this 
https://www.advantech.com/products/5754ec6c-de46-4543-ab4e-f2a8d3e96c9c/pce-usb/mod_4bcf4f15-e3ad-4cf3-809c-c49eb0dcc439