realsenseai/librealsense #4628: the depth intrinsic paramenters after depth align to color(1920*1080)

panapple, 2019-08-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  SR300 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello! I align depth(640*480) to color(1920*1080), and I wonder if I can use original depth intrinsic when I project points3d to depth .  And I also wonder how to resize (1920*1080) to (640*480) . The way I use is  cut color[:,240:1680] and resize it to (640*480).

ev-mp, 2019-08-19:
@panapple , when aligning depth frame to color the resulted depth frame is a different entity with its own size, intrinsic and extrinsic parameters. You need to use the newly-generated intrinsic to apply de-project correctly.
The second part of the question is not clear, though it seem that you're trying to somehow correlate the new "`aligned depth` frame with the original one.
Please review this scheme based on the numbers provided as it illustrates the transformation and the resulted frame type:
[image]

In addition to the camera matrices for the original and the resulted depth frames being different, the data in the aligned depth frame undergoes a `projective` transformation (i.e. each pixel is recalculated for the new camera matrix), and in spite of a great similarity between those two images, finding a correspondence cannot be achieved with a crop->resize technique.

ev-mp, 2019-09-26:
@panapple , closing as no follow ups were requested.
Feel free to reopen if I'm wrong.

closed 2019-09-26 by ev-mp