realsenseai/librealsense #4629: Face problem of face point cloud reconstruction

nevynwong, 2019-08-13:
The post-processing result of the depth image I obtained was transformed into point cloud, and the reconstructed result is shown in the following figure. 
[image]

This result is similar to the one in the previous question, and his result is like this.
[image]

When I change the depth unit to 100, the effect of a-factor to 0.18 is as follows.However, the results after post-treatment are still not satisfactory, much worse than the one without treatment below
[image]
However, these two effects are still unsatisfactory. The effect I want to achieve is the same as that in the previous question, as shown in the figure below. 
[image]

Then how can I adjust the parameters to make the effect of face point cloud good, smooth and clear.Thanks.

AntoineWefit, 2019-08-13:
I'm not sure if I understood your question correctly, but I'd say that the following result looks pretty good for an Intel RealSense :
[image]
If it's indeed what you need, you can easily smooth and close the holes in this mesh by enabling the **post-processing** filters (either with the Viewer or via the SDK).

Also, could you tell us which camera are you using and share your configuration file (.json) please ?

Now if you want to achieve maximum accuracy, you'd have to be as close as possible to the face you want to scan (look up for the min Z of your camera), at least for the D400 models.

Hope this'll help !

nevynwong, 2019-08-14:
My camera model is D415. I still have a question.If I change the depth unit to 100us, the face will be much smoother and more accurate, but the side position or the edge position of the face will be very poor, I wonder why?Do we need to change the depth unit?

AntoineWefit, 2019-08-14:
> If I change the depth unit to 100us, the face will be much smoother and more accurate, but the side position or the edge position of the face will be very poor, I wonder why?Do we need to change the depth unit?

If you need accuracy, then keep the depth unit to 100. Also, be sure to set your _Disparity Shift_ accordingly.
I think that the poor accuracy on the edges is due the IR pattern being "stretched" by the curvature of the surface. In other words, the camera is great to scan surfaces that are more or less **parallel** to it, which is the case for the face in your example, but not really for the ears area.
As you are doing a total head reconstruction with multiple scans, then I would advise you to just rotate the head and scan the "problematic" areas to get clean data.
If I'm not mistaken, the post-processing filter _Spatial Filter_ (or _Temporal Filter_ ? can't remember...) also help to get sharper edges, so maybe give it a try :)

nevynwong, 2019-08-14:
Without changing the depth unit, I made the a-factor be 0.18. The results in Meshlab are as follows. How to make the results smoother and more accurate?
[image]

AntoineWefit, 2019-08-14:
Have you tried what I suggested ?
Try to enable the _Depth to Disparity_ in the post-processing filters to get rid of this layer effect.

nevynwong, 2019-08-14:
I tried, but the result is still the same, I am still very chagrin how to solve this problem.

AntoineWefit, 2019-08-14:
Could you share your configuration file (.json) please ?

nevynwong, 2019-08-14:
In my setting, the depth unit is 100, Disparity Shift is 40, and the a-factor is 0.08. The following are the two reconstruction results.

[image]
[image]
The first one is the point cloud reconstruction directly converted by rs.convert.exe, and the second one is the point cloud reconstruction result obtained by export_to_ply after depth align to color and post-processing , I don't know whether it is the post-processing or the alignment that makes the second result very poor.

AntoineWefit, 2019-08-14:
Have you installed the [Intel RealSense Viewer](https://github.com/IntelRealSense/librealsense/releases/download/v2.25.0/Intel.RealSense.Viewer.exe) ? It is much more convenient for testing configurations. Plus you'd have a real time feedback and access to every parameters available.

nevynwong, 2019-08-15:
Thanks for your prompt reply. I will try these.

Jason420Git, 2019-08-15:
I have a similar issue to you. In what way are you scanning the face? How many frames, and in what motion? I am finding I can get quite an accurate reconstruction on areas straight on but edges like the nose at given angles end creating a lot of noise. You can try using something like TSDF to integrate multiple frames

https://github.com/intel-isl/Open3D/issues/1059

RealSenseCustomerSupport, 2019-09-04:
----------------------------------------------

@nevynwong, can we close this issue if there is no further question or update?

closed 2019-09-11 by RealSenseCustomerSupport