realsenseai/librealsense #4633: Set up to record without stopping the pipeline

g2-bernotas, 2019-08-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400   | 
| Firmware Version                   |  05.11.01.100  | 
| Operating System & Version |   Win 10  | 
| Platform                                 |  PC  |
| SDK Version                            |  2.19.0  | 
| Language                            |  C  | 
| Segment			|  others  | 

### Issue Description
I am using RealSense depth data (D435) to detect objects in the scene. As soon as it is detected by my software, I stop the streaming RealSense pipeline and enable it to record video of this object. The problem is that to stop the streaming and start the recording pipeline takes approx. 1150 ms and 300 ms, so, in total, almost 1.5s meaning that my object has almost left the FOV. 

I was wondering if there is a solution that would minimize (or even eliminate) this time taken to initialize the camera for recording.

MartyG-RealSense, 2019-08-13:
RealSense SDK 2.0 has a multi-streaming model that enables you to do different processes at the same time.  A limitation of it though is that two processes cannot access the same type of stream.  

So Process A of a camera could access the depth stream for example, whilst Process B on the same camera accesses a non-depth stream at the same time.  This implies that to detect and record at the same time, the recording and detection process must make use of different stream formats.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

g2-bernotas, 2019-08-15:
@MartyG-RealSense Thank you for your response. I may not have put all the details in my question. I am using only the depth data to detect objects, but I would like to record both colour and depth data for the 'video'. 

I have now **successfully** incorporated **recorder pause/resume** functionality for a fast recording startup. To resume the pipeline (device.as<rs2::recorder>().pause();) takes approx. 0.03 ms. As the pipeline initialization is done when the objects are not present, I am not losing any data.

closed 2019-08-15 by g2-bernotas

MartyG-RealSense, 2019-08-15:
You are very welcome.  I'm very glad you created a solution that works for you.   :)