realsenseai/librealsense #4636: Why are the pictures taken by D435 not clear when our car is moving?

Dontla, 2019-08-14:
Our D435 is mounted in a car. The car travels at a speed of 0.4m per second. When the car is still, we can get a clear picture from D435. When the car is moving and the ground is bumpy and shaking, there will be some shadows in the picture from D435. How can we eliminate these shadows and let us get the clear pictures from D435? 

We don't know if the D435 camera has a photo mode besides the video stream mode. If so, please let us know if it can improve the clarity of the image obtained from the D435 when our car is moving.

Now our video stream transmission rate of D435 camera is 30 frames per second. If we reduce the frame rate of transmission frame, can we improve the clarity of transmission image?

dorodnic, 2019-08-14:
It’s not clear from the description if you are referring to RGB or Depth data from the camera. If you can attach some pictures it would help explain the problem

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

MartyG-RealSense, 2019-08-14:
It is worth mentioning that a couple of other users have had problems with using a camera mounted to an agricultural robot or vehicle travelling on rough ground.  In those cases, they decided to explore adding vibration damping pads to the camera's mounting point.

Dontla, 2019-08-14:
@dorodnic 
Yes, I mean the RGB image. When our car is still, the image taken by D435 is as follows:
[image]
When the car is moving on a rough road, the image taken by D435 looks like this:
[image]
This makes it very difficult for us to detect objects.

MartyG-RealSense, 2019-08-14:
A while ago, someone posted a long video on YouTube of a D435 travelling on a public road with a side by side RGB and IR stream.  Though the ground is not exactly rough, you can see the vehicle undulating on the surface and making the camera move with it.

https://www.youtube.com/watch?v=OwJmCyAn3JQ

I note that whilst the D435 uses a fast global shutter for its IR, the color sensor has a slower rolling shutter.  That may make RGB not the best means to solely depend upon to do calculations in a fast-moving vehicle.

dorodnic, 2019-08-14:
You could try manually adjusting the exposure, gain and sharpness controls on the RGB sensor.
Also, you could try using grayscale infrared image (if it is reasonable for your application). IR sensor has different characteristics and may offer better results.

Dontla, 2019-08-14:
@MartyG-RealSense 
Yes, when we use the D435 camera in a vehicle with severe vibration, it's a good way to add vibration damping pads to its mounted position, but we don't know if there are any damper devices that can be recommended to me. Really thanks!

MartyG-RealSense, 2019-08-14:
Damping pads are widely available for camera tripods, so you may be able to adapt one of them.  I would recommend searching a store such as Amazon for 'tripod suppression pad'.

filipematosinov, 2019-08-21:
> You could try manually adjusting the exposure, gain and sharpness controls on the RGB sensor.
> Also, you could try using grayscale infrared image (if it is reasonable for your application). IR sensor has different characteristics and may offer better results.
@dorodnic How can we do that? I can only find how to set them using the pybackend module which we try to avoid

MartyG-RealSense, 2019-08-21:
@filipematosinov   Could you confirm the method by which you want to change exposure, gain and sharpness please?  RealSense Viewer, C++, C# or Python.  Thanks!

filipematosinov, 2019-08-21:
@MartyG-RealSense  I want to use Python

MartyG-RealSense, 2019-08-21:
If you would like to change exposure in Pyrealsense2, I hope that the script in the link below will be of help to you.

https://github.com/IntelRealSense/librealsense/issues/2683

There is another Python example here:

https://forums.intel.com/s/question/0D70P000005PM7dSAG

I'm not so sure about the gain and sharpness, though it looks as though it is accessed through the **rs.option.gain** and *rs.option.sharpness** instructions.  

https://github.com/IntelRealSense/librealsense/issues/1874#issuecomment-397185079

 I don't know about sharpness.

Dontla, 2019-09-11:
Thanks for all！I will close this issue.

closed 2019-09-11 by Dontla