realsenseai/librealsense #4640: help with reconstruction using Realsense + open3D

mukundhan3, 2019-08-14:
I need some help with reconstruction using Realsense + open3D

The steps I have followed so far:
1. I use Realsense D415 and scan a colleague's face for about 10 seconds

2. I use the Python RealSense example at https://github.com/intel-isl/Open3D/issues/473#issuecomment-508139946 to generate the PLYs from the scan (I get about 300 PLYs)

3. I use the standard open3D example at http://www.open3d.org/docs/release/tutorial/Advanced/multiway_regist
[image]
ration.html to reconstruct the 3D from the PLYs obtained in Step #2

The screenshot of the result is attached. It is quite bad. Don't know what settings I should use to improve my results. 

What I actually need a simple pipeline where my Python code would scan and generate PLYs. And then I want to take the PLYs to a CUDA enabled PC to reconstruct the full 3D object.

Have been really struggling with this. Any suggestions?
[image]

MartyG-RealSense, 2019-08-14:
Advice about improving a ply that looks poor when loaded into another program can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/2009#issuecomment-403380726

RealSenseCustomerSupport, 2019-09-13:
----------------------------------------------

Hi @mukundhan3,

Did @MartyG-RealSense's reference to https://github.com/IntelRealSense/librealsense/issues/2009 solve your problem?

Thank you!

closed 2019-09-25 by RealSenseCustomerSupport