realsenseai/librealsense #4641: T265/D435 lag while used together

vladHoloArch, 2019-08-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265/D435} | 
| Firmware Version                   | (0.0.18.5715/05.11.04.00) | 
| Operating System & Version |   Win 10| 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            |  2.23.0                       | 
| Language                            |  C#/unity                         | 
| Segment			|  AR/others                         | 

### Issue Description
Hi, when initialized like so (pretty much copied from your example) and operated by a thread executing the following operate() method there's an extreme lag received while polling which results in 16 frames received from T265 over a period of 1sec <.
If each pipeline is initialized on it's own and polled by 2 different threads simultaneously there's no lag but I keep getting StackEmpty exception. I sure can use some guidance..

        Context ctx = new Context();
        m_Pipes = new List<Pipeline>();
        var devices = ctx.QueryDevices();

        foreach (Device dev in devices)
        {
            Pipeline pipe = new Pipeline(ctx);
            Config cfg = new Config();
            cfg.EnableDevice(dev.Info.GetInfo(CameraInfo.SerialNumber));
            pipe.Start(cfg);
            m_Pipes.Add(pipe);
        }

        m_Worker = new Thread(operate);
        m_Worker.IsBackground = true;
        m_Worker.Start();

 private void operate()
    {
        while (!r_StopEvent.WaitOne(0))
        {

            FrameSet frames;
            if (m_Pipes != null)
            {
                foreach (var pipe in m_Pipes)
                {
                    if (pipe.PollForFrames(out frames))
                        using (frames)
                        {

                            var poseFrame = frames.PoseFrame;

                            if (poseFrame == null)
                            {
                                if (DepthAndColorCB != null)
                                {
                                    DepthAndColorCB.Invoke(frames);
                                }
                            }
                            else
                            {
                                if (SlamCB != null)
                                {
                                    SlamCB.Invoke(poseFrame);
                                }
                            }
                        }
                    else
                        Thread.Sleep(10);
                }
            }
        }
    }

MartyG-RealSense, 2019-08-14:
I hope the advice in the comment linked to below will be useful.

https://github.com/IntelRealSense/librealsense/issues/4163#issuecomment-500380153

vladHoloArch, 2019-08-14:
I'm trying to... but the t265 is lagging so bad there's barely any update

MartyG-RealSense, 2019-08-14:
Does it help if you change the thread.sleep value from '10' to '1' ?

vladHoloArch, 2019-08-14:
I did, it makes no difference. I've tried creating the pipelines in different binaries and polling them T265 first in a different binary and that doesn't never retrieves any frames..Im clueless, it seems the best course is keep on working in 2 threads and manage the exception.

RealSenseCustomerSupport, 2019-09-06:
----------------------------------------------

@vladHoloArch - I see that MartyG proposed the thread outlined in https://github.com/IntelRealSense/librealsense/issues/4163, I'd also add the threads discussed in these two as well - https://github.com/IntelRealSense/librealsense/issues/4709, and https://github.com/IntelRealSense/librealsense/pull/4302 if you haven't read through them already.

closed 2019-09-18 by RealSenseCustomerSupport