realsenseai/librealsense #4642: error when restart pipeline

xhuan28, 2019-08-15:
### Issue Description
I would like to implement a function that user can stop pipeline and restart it at the runtime.
Below is my logic:
```
if (condition 1)  
{  
  pipeline.stop();  
}  
if (condition 2)  
{  
  config.enable_stream(stream_name, stream_index);  
  pipeline.start(config);  
}
```
There is no problem when I was trying to stop pipleine and restart color stream, depth stream and infra stream. However, when I was trying to do this with T265, error occurred.
1. restart fisheye

> terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  Invalid profile configuration - setting a single FE stream is not supported

2. restart accel and gyro

> 13:41:40.794 [24536] [E] Device-3900: MessageID 0x5 (DEV_RAW_STREAMS_CONTROL) failed with status 0x8
13:41:40.794 [24527] [E] Device-3900: Error: Failed setting Supported Raw Streams (0x1)
terminate called after throwing an instance of 'rs2::error'
  what():  Failed to start TM2 camera

3. restart pose

> 13:42:06.620 [24575] [E] Device-3900: MessageID 0x1006 (SLAM_6DOF_CONTROL) failed with status 0x8
13:42:06.620 [24566] [E] Device-3900: Error Transferring SLAM_6DOF_CONTROL
13:42:06.620 [24566] [E] Device-3900: Error: Failed setting 6dof Control (0x6)
terminate called after throwing an instance of 'rs2::error'
  what():  Failed to start TM2 camera

Is there any constraint in T265? Can anyone give me some hints?

jebasamuel, 2019-08-16:
@xhuan28 
The error code 0x8 means LIBUSB_ERROR_OVERFLOW.

I had the same situation.When delay is introduced between stop() and start(), it looks OK.

pipe.stop()
<delay.. For example, 200ms>
pipe.start()

jebasamuel, 2019-08-16:
Stopping the pipe
18:39:37.774 [29336] [E] Device-9690: Sent request - MessageID: 0x13 (DEV_STOP), Len: 6, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:37.777 [29336] [E] Device-9690: Got response - MessageID: 0x13 (DEV_STOP), Len: 8, Status: 0x0, UsbLen: 8, Actual: 8, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
Restarting the pipe
18:39:38.145 [29333] [E] Device-9690: Sent request - MessageID: 0x1 (DEV_GET_DEVICE_INFO), Len: 6, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.145 [29333] [E] Device-9690: Got response - MessageID: 0x1 (DEV_GET_DEVICE_INFO), Len: 54, Status: 0x0, UsbLen: 54, Actual: 54, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.147 [29336] [E] Device-9690: Sent request - MessageID: 0x18 (DEV_GET_TEMPERATURE), Len: 6, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.149 [29336] [E] Device-9690: Got response - MessageID: 0x18 (DEV_GET_TEMPERATURE), Len: 48, Status: 0x0, UsbLen: 1024, Actual: 48, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.149 [29336] [E] Device-9690: Sent request - MessageID: 0x18 (DEV_GET_TEMPERATURE), Len: 6, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.150 [29336] [E] Device-9690: Got response - MessageID: 0x18 (DEV_GET_TEMPERATURE), Len: 48, Status: 0x0, UsbLen: 1024, Actual: 48, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.150 [29333] [E] Device-9690: Sent request - MessageID: 0x1 (DEV_GET_DEVICE_INFO), Len: 6, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.150 [29333] [E] Device-9690: Got response - MessageID: 0x1 (DEV_GET_DEVICE_INFO), Len: 54, Status: 0x0, UsbLen: 54, Actual: 54, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.151 [29336] [E] Device-9690: Sent request - MessageID: 0x6 (DEV_GET_CAMERA_INTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.151 [29336] [E] Device-9690: Got response - MessageID: 0x6 (DEV_GET_CAMERA_INTRINSICS), Len: 56, Status: 0x0, UsbLen: 56, Actual: 56, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.152 [29336] [E] Device-9690: Sent request - MessageID: 0x6 (DEV_GET_CAMERA_INTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.152 [29336] [E] Device-9690: Got response - MessageID: 0x6 (DEV_GET_CAMERA_INTRINSICS), Len: 56, Status: 0x0, UsbLen: 56, Actual: 56, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.153 [29336] [E] Device-9690: Sent request - MessageID: 0x7 (DEV_GET_MOTION_INTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.153 [29336] [E] Device-9690: Got response - MessageID: 0x7 (DEV_GET_MOTION_INTRINSICS), Len: 80, Status: 0x0, UsbLen: 80, Actual: 80, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.156 [29336] [E] Device-9690: Sent request - MessageID: 0x7 (DEV_GET_MOTION_INTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.156 [29336] [E] Device-9690: Got response - MessageID: 0x7 (DEV_GET_MOTION_INTRINSICS), Len: 80, Status: 0x0, UsbLen: 80, Actual: 80, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.160 [29336] [E] Device-9690: Sent request - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.160 [29336] [E] Device-9690: Got response - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 57, Status: 0x0, UsbLen: 57, Actual: 57, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.162 [29336] [E] Device-9690: Sent request - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.162 [29336] [E] Device-9690: Got response - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 57, Status: 0x0, UsbLen: 57, Actual: 57, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.163 [29336] [E] Device-9690: Sent request - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.163 [29336] [E] Device-9690: Got response - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 57, Status: 0x0, UsbLen: 57, Actual: 57, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.164 [29336] [E] Device-9690: Sent request - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.164 [29336] [E] Device-9690: Got response - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 57, Status: 0x0, UsbLen: 57, Actual: 57, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.164 [29336] [E] Device-9690: Sent request - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 7, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.164 [29336] [E] Device-9690: Got response - MessageID: 0x8 (DEV_GET_EXTRINSICS), Len: 57, Status: 0x0, UsbLen: 57, Actual: 57, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.191 [29336] [E] Device-9690: Sent request - MessageID: 0x1006 (SLAM_6DOF_CONTROL), Len: 8, UsbLen: 1024, Actual: 1024, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.191 [29336] [E] Device-9690: Got response - MessageID: 0x1006 (SLAM_6DOF_CONTROL), Len: 8, **Status: 0x8**, UsbLen: 8, Actual: 8, rc: 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED)
18:39:38.191 [29336] [E] Device-9690: MessageID 0x1006 (SLAM_6DOF_CONTROL) failed with status 0x8
18:39:38.191 [29333] [E] Device-9690: Error Transferring SLAM_6DOF_CONTROL
18:39:38.191 [29333] [E] Device-9690: Error: Failed setting 6dof Control (0x6)
RealSense error calling rs2_pipeline_start_with_config(pipe:0x5c10b0, config:0x5b11c0):
    Failed to start TM2 camera

xhuan28, 2019-08-19:
@jebasamuel 
Thanks a lot for your answers. As you suggested, the error doesn't exist when I make it sleep 200ms between stop() and start().

closed 2019-08-19 by xhuan28