realsenseai/librealsense #4643: rs2.rs2_ deproject_pixel_to_point not giving values in X and Y coordinates in gazebo simulation

KapilRawal19, 2019-08-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (v2.25.0) | 
| Operating System & Version |   Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  (4.15.0-55-generic)                                         | 
| Platform                                 | PC
| SDK Version                            |   legacy / 2.<?>.<?>                           | 
| Language                            |  python                           | 
| Segment			|  Drone                          | 

### Issue Description
Hello,
I used a D435 for measuring the distance from an object using rs.rs2_deproject_pixel_to_point, it gave me accurate coordinates of the object in the world frame. I had subscribed to depth topic '/camera/color/camera_info', '/camera/aligned_depth_to_color/image_raw'.  

Now when I'm trying to replicate the same real-world scenario in gazebo simulation. In gazebo simulation, i found firefly/rotor-simulator has the package for the stereo camera so I'm using that for depth measurement(It publishes rgb8 images and L8 depth images format ). I have subscribed to the depth topic and camera info of stereo. I checked on rviz that I'm getting an accurate depth from the simulation. Now when i use rs.rs2_deproject_pixel_to_point it doesn't give me values near to where the object is in the gazebo. Take, for example, I have the stereo camera at [0,0,1] in the gazebo and the object at [1,0,1]. Now the expected output of the rs.rs2_deproject_pixel_to_point(<from camera info>, pixel points are known, depth in meters from depth topic) would be around the coordinates [1,0,1] but the actual output is something like [0.004,0.002,1.738]. The values in X and Y coordinates doesn't seem to change from 0. 

So my question is where I'm going wrong for an object position calculation, is the issue due to subscribing to normal depth topic instead of aligned depth topic?. 
Let me know if more info is required for understanding the question.

Thank you for your time!

RealSenseCustomerSupport, 2019-12-06:
----------------------------------------------

@KapilRawal19 Sorry for late response. Did you still have this issue? Did you try subscribing aligned depth topic to see if still have issue?

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

@KapilRawal19 Sorry that we didn't get your response for weeks. Will close it at this point. Please feel free to create another one if you still have questions or issues. Thanks!

closed 2019-12-23 by RealSenseCustomerSupport