realsenseai/librealsense #4644: Alignment creates some offset

nevynwong, 2019-08-15:
| Camera Model                       |D415 | 
| Operating System & Version |  Linux (Ubuntu16） | 
| Language                            | python  | 

After depth alignment to color and post-processing, the resulting point cloud is slightly skewed, like the horizontal and vertical grid lines in the image below. How to avoid this?I compared the ply that was converted by rs-convert.exe and there were no such lines
[image]

ev-mp, 2019-08-19:
@nevynwong hi,
Creating a point-cloud from `Depth aligned to color` generates a slightly distorted 3D model compared to the point-cloud model from the original depth frame. This is anticipated as during the transformations the calibration and aliasing tolerances accumulate in the aligned image, resulting in measurable/visible artifacts observed when the depth is de-projected back into 3D space.
The level of biases depends both on hardware (optical path) and software (calibration, resolution) characteristics.
These artifacts cannot be easily removed, though there are certain ways to mitigate it to some extent:
 - Use Full HD for RGB sensor and the recommended resolution for Depth sensor (1280X720 for D415 and 848X480 for D435(i))
- Enhance the calibration using "DynamicCalibrator" tool.

Eventually, if you need to use both the aligned depth (2D processing) and the best-quality point-cloud data for 3D processing/rendering, I would recommend to generate and store both the original and "Aligned" depth frames and use the latter for the point-cloud generation.

Previously discussed in several threads, e.g. #4536, #4315, #2523

ev-mp, 2019-09-02:
@nevynwong , was it useful ? 
Looking forward for an update.

RealSenseCustomerSupport, 2019-09-25:
----------------------------------------------

Hi @nevynwong, please let us know if further assistance is needed on this topic. Thank you.

closed 2019-10-02 by RealSenseCustomerSupport