realsenseai/librealsense #4646: YUY2 format to rgb format

waiyc, 2019-08-15:
Hi,

Is there anyways to convert the YUY2 format from RGB camera to rgb format using opencv.
```
config.enable_stream(rs.stream.color, 1920, 1080, rs.format.yuyv, 15)
...
color_frame = frames.get_color_frame()
yuyv_image = np.asanyarray(color_frame.get_data())
rgb_image = cv2.cvtColor(yuyv_image, cv2.COLOR_YUV2BGR_YUY2)
```
I have tried using `cv2,cvtColor()` with `COLOR_YUV2BGR_YUY2` but failed to convert.
> error: (-215) scn == 2 && depth == 0 in function cvtColor

MartyG-RealSense, 2019-08-15:
I wonder if you might be able to adapt the conversion information in the link below.

https://answers.opencv.org/question/87083/yuyv422-to-bgr/?answer=202592#post-id-202592

waiyc, 2019-08-15:
@MartyG-RealSense  thanks for the reply.
I have tried the same ColorConversionCodes mentioned in the discussion, but still facing the same error.

MartyG-RealSense, 2019-08-15:
Apologies for the delay in responding further.  I was carefully researching your error message.  I came across this case with the same error:

https://stackanswers.net/questions/error-trying-to-use-cvtcolor-with-cv2-color-yuv2bgr-y422-error-215-scn-2-depth-0-in-function-cv-cvtcolor

From what I can gather (I am not an OpenCV expert), the scn==2 error may occur because the convert function needs an image with 2 channels.

waiyc, 2019-08-16:
yes, in order to convert YUY2 to rgb the source image is required to contains 2 channels. 

However, the input array size after reading the rgb image with input YUYV format contains a single channel 2d array only.

```
config.enable_stream(rs.stream.color, D435_RAW_COLOR_W, D435_RAW_COLOR_H, rs.format.yuyv, D435_RAW_COLOR_FPS)
color_frame = frames.get_color_frame()
yuyv_image = np.asanyarray(color_frame.get_data())
```

yuyv_image array size = `(1080, 1920)`
yuyv_image array data type = `uint16`

---

i also came across [show_yuv_frame()](https://github.com/toinsson/pyrealsense/blob/master/examples/show_jupyter.ipynb) developed few years ago shows that `cv2.cvtColor(frame, cv2.COLOR_YUV2RGB_YUYV)` can be used to convert YUY2 format to rgb without issues previously and the color_frame array should have 2 channels.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@waiyc Sorry for later response. Did you get it through now? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-01-15:
----------------------------------------------

@waiyc Will close the issue if no other questions. Thanks!

closed 2020-01-15 by RealSenseCustomerSupport

tschoepflin, 2020-12-23:
Here's a solution...  the trick is that YUY2 format needs to be put into Y (channel 0) and UV (channel 1).  Moving forward I will never use YUY2 with OpenCV since it is heavily biased toward BGR.

```
# Read BAG file and store image

import pyrealsense2 as rs
import numpy as np
import cv2
import sys, getopt

def GetBGR(frame_color):
    # Input: Intel handle to 16-bit YU/YV data
    # Output: BGR8
    H = frame_color.get_height()
    W = frame_color.get_width()
    Y = np.frombuffer(frame_color.get_data(), dtype=np.uint8)[0::2].reshape(H,W)
    UV = np.frombuffer(frame_color.get_data(), dtype=np.uint8)[1::2].reshape(H,W)
    YUV =  np.zeros((H,W,2), 'uint8')
    YUV[:,:,0] = Y
    YUV[:,:,1] = UV
    BGR = cv2.cvtColor(YUV,cv2.COLOR_YUV2BGR_YUYV)
    return BGR

# Configure depth and color streams
config = rs.config()
pipeline = rs.pipeline()
rs.config.enable_device_from_file(config, 'g_cam0.bag')
config.enable_stream(rs.stream.color, format=rs.format.yuyv, framerate=15)

profile = pipeline.start(config)

try:
    # Get 5 images
    for i in range(60):
        frames = pipeline.wait_for_frames()
        frame_color = frames.get_color_frame()
        if frame_color:
            BGR = GetBGR(frame_color)
            cv2.imwrite(f'RS_RGB_img_{i}.jpg', BGR)

finally:
    # Stop streaming
    pipeline.stop()

MartyG-RealSense, 2020-12-23:
Thanks so much for sharing your solution with the RealSense community @tschoepflin    :)

MartyG-RealSense, 2025-06-03:
Thanks so much @xperroni for sharing your knowledge with the RealSense community!