realsenseai/librealsense #4647: How rs.disparity_transform work?

MCDM2018, 2019-08-15:
* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | latest version | 
| Operating System & Version |   {Win 10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python }                          | 
| Segment			|  AR }                          | 

### Issue Description

Hi. I m developing a 3D position tracking application.
So i need to use depth information accurately , a few month ago i referred to this script. https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

But actually the region which should be filled could be determined by external information.
So, i want to reduce the cost to fill depth holes. 

Followings are my questions . 

1. How do pyrealsense API's function 'rs.disparity_transform' work ?
And... in the end... can i replace the function with more efficiently customed function ?
2. Is there a method to modify frame data directly? ( Not using following method, 
depth = np.asanyarray(depth_frame_.get_data()) )

lramati, 2019-08-21:
The source code for the disparity_transform (In C++) can be found [here](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/disparity-transform.h)
As for modifying frame data directly, not quite. Frames are read-only, however, it is possible to define a custom filter which, with the help of `rs.frame_source` can allow you to perform processing on a realsense frame and then package the result in a new realsense frame object

ev-mp, 2019-09-02:
@MCDM2018, do you need more assistance here?

MCDM2018, 2019-09-04:
@lramati Thanks to your help
@ev-mp No thanks. Thanks for consideration

closed 2019-09-04 by MCDM2018