realsenseai/librealsense #4648: T265 android supports

vladHoloArch, 2019-08-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265} | 
| Firmware Version                   | (0.0.18.5715) | 
| Operating System & Version |   Win 10| 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC/Android |
| SDK Version                            |  2.23.0                       | 
| Language                            |  C#/unity                        | 
| Segment			|  AR/others                         | 

### Issue Description
Hi,
just wanted to make sure. There's no way to access the T265 device via librealsense aar on android right? I've read the release notes...

dorodnic, 2019-08-15:
Hi @vladHoloArch 
Not yet, unfortunately. We do plan to add it later this year, based on customer feedback / demand.

vladHoloArch, 2019-08-20:
Hi @dorodnic 
Been struggling for the last few days to write my own "driver" in java.
The android device recognizes the Movidius ma215x and it seems like a connection can be established. 
But when I try to read data from the UsbEndpoint-out I receive nothing..
I know I need to establish a connection with some kind of write bulkBufferTransfer command to the 0 endpoint.
I can't find any official or non official datasheet documentation about the movidius chip usb interface and the T265 datasheet is lean to say the least..
I would really appreciate some direction or guidance cause I'm clueless...

radfordi, 2019-08-21:
Hi @vladHoloArch,

You need to uploaded (with a single bulk write) the firmware (`target.mvcmd`) to the Movidius device before it turns into a T265 device from which you can get data.  After that, you can start the pose or image streams.

vladHoloArch, 2019-08-21:
Thanks @radfordi 
do I just convert the firmware to bytes and send it to endpoint 0? there's no api how to I format my request to open pose/image stream?

radfordi, 2019-08-21:
Yes.
```C++
static const int USB_TIMEOUT = 10000/*ms*/;

static void tm2_boot(libusb_device *device, const std::string &fw) {
    libusb_device_handle *handle = nullptr;
    if (libusb_open(device,&handle))
        return;
    if (libusb_claim_interface(handle, 0))
        goto close;
    {
        int written = -1;
        int e = libusb_bulk_transfer(handle, 0x1, (unsigned char*)fw.c_str(), fw.size(), &written, USB_TIMEOUT);
        if (e || written != (int)fw.size())
            std::cerr << "Cannot write firmware to TM2 bootloader " << written << "/" << fw.size() << " bytes written: " << libusb_error_name(e) << "\n";
    }

    libusb_release_interface(handle, 0);
  close:
    libusb_close(handle);
}
```

vladHoloArch, 2019-08-21:
@radfordi can you show me what you use for opening the stream after writing the firmware?
does the device needs to be requested each time before it writes the frames to endpoint 1 or it just starts writing and overwriting the last data? meaning do I just need to read from the endpoint after opening the stream?

RealSenseCustomerSupport, 2019-09-18:
----------------------------------------------

@vladHoloArch, Is there anything further required on this ticket?

jabrena, 2019-09-22:
Hi @vladHoloArch, @radfordi,

any progress in the POC to connect from Android with T265?

Many thanks in advance

Juan Antonio

vladHoloArch, 2019-09-25:
@jabrena 
yep, check out my repo - [https://github.com/vladHoloArch/SlamDriver/blob/master/t265lib/src/main/java/com/example/t265lib/USBController.java](here)

RealSenseCustomerSupport, 2019-09-26:
----------------------------------------------

With the update from @vladHoloArch, can this be closed out?

Thanks

closed 2019-10-04 by RealSenseCustomerSupport

radfordi, 2019-11-08:
Hi @vladHoloArch, this PR #5213, replacing `libtm`, drastically simplifies the `librealsense` interface to the T265, and should give you some ideas for your Android port.

TheBricktop, 2019-11-18:
This issue shouldnt be closed as there still is no official support for t265 on Android.

PapaFuchs, 2019-11-22:
I agree with @TheBricktop. As long as there is no official support this thread would be really useful to get some information about when we can expect it to work.

@vladHoloArch is your repo/code still accessible somewhere?

neilyoung, 2019-11-22:
> Hi @vladHoloArch, this PR #5213, replacing `libtm`, drastically simplifies the `librealsense` interface to the T265, and should give you some ideas for your Android port.

This is true, it makes USB usable for the first time on macOS too. But as a side effect it makes python unusable. One should know. See https://github.com/IntelRealSense/librealsense/issues/5276#issuecomment-555743637 please

neilyoung, 2019-11-22:
> > Hi @vladHoloArch, this PR #5213, replacing `libtm`, drastically simplifies the `librealsense` interface to the T265, and should give you some ideas for your Android port.
> 
> This is true, it makes USB usable for the first time on macOS too. But as a side effect it makes python unusable. One should know. See [#5276 (comment)](https://github.com/IntelRealSense/librealsense/issues/5276#issuecomment-555743637) please

Hmm. Maybe you disregard that. Seems, the development branch has a general issue with python, regardless of this PR