realsenseai/librealsense #4649: linking statically Win

Armin234, 2019-08-16:
Hi,

where i can get a static lib  librealsense2.lib do linking with my program?

Armin

MartyG-RealSense, 2019-08-16:
The CMake customization flags for building Librealsense say that if the flag **BUILD_SHARED_LIBS** is set to **False** (it is True by default) then the output will be a static library instead of a dynamic link library (DLL) or a shared object (SO).

https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration

Armin234, 2019-08-16:
okay, that means I have to download the entire package and try to compile it myself …  :-(

dorodnic, 2019-08-19:
That is true.
Let us know if you encounter any problems building the package with CMake and Visual Studio. 
Static version is not suitable for some applications (for instance .NET P/Invoke) but I agree it would be useful to bundle the full static library in the installer, in addition to what we have right now.

Armin234, 2019-08-20:
By copying the DLL into the Windows System32 directory, my problem has been solved, even if it is the sloppiest solution.
I was irrated that all precompiled programs in the SDK / Tools worked until I understood that they were linked statically and not dynamically like the project examples.

Personally, I prefer the static linking, probably for the same reasons why the tools in the SDK were linked statically.

For me, there are several approaches:
- The installation of the SDK copies the DLL into appropriate directories, so that it is also found
- the installation of the SDK sets the appropriate environment variables, so that this is also found
- The static library is provided and also the examples adapted
- If the dynamic variant is preferred, a redistributable package should also be provided.

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

This change request is approved. We will implement it.

MartyG-RealSense, 2022-04-23:
Hi @Armin234  I would like to provide an update about the feature request in this case for static linking in Windows.  The feature request created by Intel remains active but there is no progress to report at the time of writing this.

MartyG-RealSense, 2026-10-29:
Closing this case due to the feature request being closed by RealSense as a won't do in December 2024.

closed 2026-10-29 by MartyG-RealSense