realsenseai/librealsense #4651: Crash  rs2_delete_stream_profile

Armin234, 2019-08-16:
Hi, 
when I open a stream and close it without getting any frame, my program chrash at the rs2_delete_stream_profile?

Armin

#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_option.h>
#include <librealsense2/h/rs_frame.h>
// #include <librealsense2/rsutil.h>

#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

//                                     These parameters are reconfigurable                                        //

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

#define STREAM          RS2_STREAM_DEPTH  // rs2_stream is a types of data provided by RealSense device           //

#define FORMAT          RS2_FORMAT_Z16    // rs2_format is identifies how binary data is encoded within a frame   //

#define WIDTH           0               // Defines the number of columns for each frame or zero for auto resolve//

#define HEIGHT          0                 // Defines the number of lines for each frame or zero for auto resolve  //

#define FPS             30                // Defines the rate of frames per second                                //

#define STREAM_INDEX    0                 // Defines the stream index, used for multiple streams of the same type //

#define HEIGHT_RATIO    20                // Defines the height ratio between the original frame to the new frame //

#define WIDTH_RATIO     10                // Defines the width ratio between the original frame to the new frame  //

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

void check_error(rs2_error* e)
{
	if (e)
	{
		printf("rs_error was raised when calling %s(%s):\n", rs2_get_failed_function(e), rs2_get_failed_args(e));
		printf("    %s\n", rs2_get_error_message(e));
		exit(1);
	}
}

// The number of meters represented by a single depth unit

float get_depth_unit_value(const rs2_device* const dev)
{
	rs2_error* e = 0;
	rs2_sensor_list* sensor_list = rs2_query_sensors(dev, &e);

	check_error(e);

	int num_of_sensors = rs2_get_sensors_count(sensor_list, &e);
	check_error(e);

	float depth_scale = 0;

	int is_depth_sensor_found = 0;

	int i;

	for (i = 0; i < num_of_sensors; ++i)
	{
		rs2_sensor* sensor = rs2_create_sensor(sensor_list, i, &e);
		check_error(e);
		// Check if the given sensor can be extended to depth sensor interface

		is_depth_sensor_found = rs2_is_sensor_extendable_to(sensor, RS2_EXTENSION_DEPTH_SENSOR, &e);
		check_error(e);

		if (1 == is_depth_sensor_found)
		{
			depth_scale = rs2_get_option((const rs2_options*)sensor, RS2_OPTION_DEPTH_UNITS, &e);
			check_error(e);
			rs2_delete_sensor(sensor);
			break;
		}
		rs2_delete_sensor(sensor);
	}
	rs2_delete_sensor_list(sensor_list);

	if (0 == is_depth_sensor_found)
	{
		printf("Depth sensor not found!\n");
		exit(EXIT_FAILURE);
	}
	return depth_scale;
}

int main()

{

	rs2_error* e = 0;

	rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
	check_error(e);

	rs2_device_list* device_list = rs2_query_devices(ctx, &e);
	check_error(e);

	int dev_count = rs2_get_device_count(device_list, &e);
	check_error(e);

	printf("There are %d connected RealSense devices.\n", dev_count);

	if (0 == dev_count)
		return EXIT_FAILURE;

	rs2_device* dev = rs2_create_device(device_list, 0, &e);
	check_error(e);

	//print_device_info(dev);

	/* Determine depth value corresponding to one meter */

	float lDepthScale = get_depth_unit_value(dev);

	uint16_t one_meter = (uint16_t)(1.0f / lDepthScale);

	rs2_pipeline* pipeline = rs2_create_pipeline(ctx, &e);
	check_error(e);

	rs2_config* config = rs2_create_config(&e);
	check_error(e);

	rs2_config_enable_stream(config, STREAM, STREAM_INDEX, WIDTH, HEIGHT, FORMAT, FPS, &e);
	check_error(e);

	rs2_pipeline_profile* pipeline_profile = rs2_pipeline_start_with_config(pipeline, config, &e);
	if (e)
	{
		printf("The connected device doesn't support depth streaming!\n");
		exit(EXIT_FAILURE);
	}

	rs2_stream_profile_list* stream_profile_list = rs2_pipeline_profile_get_streams(pipeline_profile, &e);
	if (e)
	{
		printf("Failed to create stream profile list!\n");
		exit(EXIT_FAILURE);
	}

	const rs2_stream_profile* stream_profile = rs2_get_stream_profile(stream_profile_list, 0, &e);
	if (e)
	{
		printf("Failed to create stream profile!\n");
		exit(EXIT_FAILURE);
	}

	///////////////////////////////////////////////////
	rs2_intrinsics intrinsics;
	rs2_get_video_stream_intrinsics(stream_profile, &intrinsics, &e);
	if (e)
	{
		printf("Trs2_get_video_stream_intrinsics!\n");
		exit(EXIT_FAILURE);
	}
	///////////////////////////////////////////////////

	rs2_stream stream; rs2_format format; int index; int unique_id; int framerate;
	rs2_get_stream_profile_data(stream_profile, &stream, &format, &index, &unique_id, &framerate, &e);
	if (e)
	{
		printf("Failed to get stream profile data!\n");
		exit(EXIT_FAILURE);
	}

	int width; int height;

	rs2_get_video_stream_resolution(stream_profile, &width, &height, &e);
	if (e)
	{
		printf("Failed to get video stream resolution data!\n");
		exit(EXIT_FAILURE);
	}

	// Stop the pipeline streaming

	rs2_pipeline_stop(pipeline, &e);
	check_error(e);

	// Release resources

	rs2_delete_pipeline_profile(pipeline_profile);
	rs2_delete_stream_profiles_list(stream_profile_list);
	rs2_delete_stream_profile( (rs2_stream_profile*) stream_profile);
	rs2_delete_config(config);
	rs2_delete_pipeline(pipeline);
	rs2_delete_device(dev);
	rs2_delete_device_list(device_list);
	rs2_delete_context(ctx);

	return EXIT_SUCCESS;

dorodnic, 2019-08-18:
Hi @Armin234 
Solution is rather simple - don't call `rs2_delete_stream_profile` :)
It might be not entirely obvious from the documentation but `rs2_clone_stream_profile` states:
```
* \return                new stream profile, must be deleted by rs2_delete_stream_profile
```
But `rs2_get_stream_profile` says nothing about delete and only states:
```
* Get pointer to specific stream profile
```
The point is - when you define new stream profile in your application, you must delete it using `rs2_delete_stream_profile` but when you get one from a frame / profiles list, its lifetime is managed by its owner object.

dorodnic, 2019-08-18:
Also, there is a handy CMake flag called `TRACE_API` that lets you dump all the C API calls any SDK application invokes. This is often useful to understand C ownership model, since I must admit the API was designed with mainly C++ in mind.

closed 2019-08-20 by Armin234