realsenseai/librealsense #4652: Firmware update tool issues (ubuntu linux 18.03 LTS)

muralikbk, 2019-08-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version  | Linux (Ubuntu 18.04.3 LTS)   | 
| Kernel Version (Linux Only)   |  5.0.0-23-generic                                        | 
| Platform                                 | PC  |
| SDK Version                          | 2                  | 
| Language                              |  python                | 

### Issue Description
We are developing an application using the realsense D415 camera; we would prefer to update the firmware ourselves via a docker to control which version gets updated

We are using the rs-fw-update tool; whenever we try to update the firmware via the command "rs-fw-update -s <serial> -f Signed_Image_UVC_5_11_6_250.bin", we get the error:
"ERROR [140607827945152] (types.h:243) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error". This has happened in multiple docker containers so far. 

We don't see this issue outside the container - the firmware update happens smoothly. Furthermore, once the camera goes into recovery mode after the failed update, the command "rs-fw-update -r -f Signed_Image_UVC_5_11_6_250.bin" from within the docker container successfully recovers the camera using the firmware binary file and we are able to continue using the camera.

Can we get some assistance on this? Feel free to reach out for any questions.

MartyG-RealSense, 2019-08-16:
Comparing the instruction you used to the examples on the documentation for rs-firmware-update, I notice that you did not put a camera serial number after -s .... for example:

rs-fw-update -s **725112060411** -f Signed_Image_UVC_5_11_6_250.bin

If you only have one camera connected then you can leave out -s, the documentation says, and do:

rs-fw-update -f Signed_Image_UVC_5_11_6_250.bin

muralikbk, 2019-08-16:
Hi Marty,

     Thanks for the quick response - I did indeed put the serial number after the -s in the command. I didn't want to post the actual S/N of the device (since it belongs to the company) so I added a placeholder \<serial\> that seems to have been edited out when displaying.

     The issue isn't that the firmware update doesn't occur - the same command works when I execute it outside the docker container. From inside the docker container however, the command fails with the error I posted above.

MartyG-RealSense, 2019-08-16:
You mentioned that you were able to execute camera recovery from within the docker with -r, which suggests that there is some form of successful connectivity between the camera and docker (at least for that particular type of instruction).

Does the docker also fail if you leave -s out and only try an -f update for a single camera?  Or are you unable to detach other cameras you may have connected in order to test it this way?  Thanks!

muralikbk, 2019-08-16:
I just tried it out. Same exact error message.

MartyG-RealSense, 2019-08-16:
In a way, the documentation for the -f instruction (for the path) does not make logical sense to me, as there is nothing after -f (unlike -s, where a serial number has to follow afterwards).  I wonder if the path would be different inside the docker and outside of it.  

I can imagine a scenario where just having -f on its own without a path causes it to just look for the firmware bin file in the same location that the rs-fw-update tool was executed.  I do not know that for sure though, as I am not familiar with the workings of the new cross-platform firmware tool.

muralikbk, 2019-08-16:
The firmware update utility tries to search for the file first - if the file isn't found, it doesn't even start the update process.
In this case, the update process does get started and is somehow interrupted - this is why the camera goes into recovery mode. The same file path works in the recover mode, so the file location doesn't seem to be the issue.

MartyG-RealSense, 2019-08-16:
If it is not the serial number (because it still occurs if you leave -s out) and it is not the path, and the firmware version name is correct, it does not leave much to go wrong in your execution instruction.  And updating the firmware successfully with -r by making it fail first isn't an ideal workaround.   :)

Any chance that a problem is being caused by using "--privileged" and "-v /dev/bus/usb:/dev/bus/usb" ?

dorodnic, 2019-08-18:
Hi @muralikbk 
This seem rather specific to Video4Linux behavior inside docker. I would recommend to try building the tool with `-DFORCE_LIBUVC=true` CMake flag.

muralikbk, 2019-08-19:
Hi Marty,
    The privileged and usb mapping is required for the container to access the camera. I am not sure if the rs-fw-update tool itself requires additional rights; hence requesting assistance.

Hi Dorodnic,
     Not sure how Video4Linux plays into this - I got this tool by doing apt-get for the utils in Ubuntu. I did not compile it from source and am not sure how feasible the compilation action is from within a docker container.

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@muralikbk Any update? Thanks!

muralikbk, 2019-12-02:
Hi - Apologies for not responding sooner. I am no longer working at the position that required this of me - The company has opened a separate case with Intel for this and will follow up on that thread. Feel free to close this one.

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@muralikbk Thanks for the update! Close this.

muralikbk, 2019-12-13:
Closing per comments.

closed 2019-12-13 by muralikbk