realsenseai/librealsense #4655: D435 Depth Module: Restart Issue / No Frames

dgo-pi, 2019-08-16:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 (depth module only) | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |  Yocto Linux  | 
| Kernel Version (Linux Only)    |  4.9.88   | 
| Platform                                 | Arm 64 NXP IMX 8m quad core  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  C++                          | 
| Compiler                            |  g++ 7.3.0                          | 
### Issue Description
D435 Hangs On Restart.

Thank you for looking at this.
Yes, we did look at the other related issues, but this seems different than previously-submitted issues.

The initial run is completely nominal. We stop the camera (using pipeline.stop() method) before exiting. On the second run of the program without unplugging / re-plugging camera, this issue appears. Another thing that we have noticed that during this error state in the second run, the IR projector is not operating; indicating that the camera may not be recording at all.

This issue only seems to present on the ARM platform; and this issue is **not** apparent on the PC platform under Linux.

Error message is:
```
terminate called after throwing an instance of 'rs2::error'
  what():  Frame didn't arrived within 15000
Aborted (core dumped)
```

Attempting to unwind the generated core file using gdb results in:
```
Core was generated by `./LuciSensors -s ../../configs/sensors.toml'.
Program terminated with signal SIGABRT, Aborted.
#0  0x0000ffff787bb88c in raise () from /lib/libc.so.6
[Current thread is 1 (LWP 3410)]
(gdb) bt full
#0  0x0000ffff787bb88c in raise () from /lib/libc.so.6
No symbol table info available.
#1  0x0000ffff787bcb8c in abort () from /lib/libc.so.6
No symbol table info available.
#2  0x0000ffff78a45b08 in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/libstdc++.so.6
No symbol table info available.
#3  0x0000ffff78a436e4 in ?? () from /usr/lib/libstdc++.so.6
No symbol table info available.
#4  0x0000ffff78a43730 in std::terminate() () from /usr/lib/libstdc++.so.6
No symbol table info available.
#5  0x0000ffff78a6e3f0 in ?? () from /usr/lib/libstdc++.so.6
No symbol table info available.
#6  0x0000ffff79208f40 in ?? () from /lib/libpthread.so.0
No symbol table info available.
#7  0x0000ffff775a2688 in ?? ()
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 

```

All is nominal on the first run. 
`pipeline.stop()`
is invoked when we want to stop the camera, and before we exit.

We suspect that something is going wrong deep within librealsense, even to the point in corrupting the stack as per the gdb message above.

This is being submitted from the same team which submitted
https://github.com/IntelRealSense/librealsense/issues/4608
and is using the same code fragment as presented there.

dgo-pi, 2019-08-19:
An additional piece of information:

Prior to the failure listed above, during the warm start the USB connection ***is*** working, as the camera data is retrievable, so we believe that this is not a USB issue; and may be within the RealSense stack or the camera firmware. The fragment below is right before the "terminate" terminal dump above:

```
Name: Intel RealSense D435
Serial #: 819112071223
  Installed Firmware: 05.11.11.100
Recommended Firmware: 05.11.06.250
Phys Port: /sys/devices/platform/usb@38200000/38200000.dwc3/xhci-hcd.1.auto/usb4/4-1/4-1.4/4-1.4:1.0/video4linux/video0
Advanced Mode: YES
Product ID: 0B07
USB Type Descriptor: 3.2

terminate called after throwing an instance of 'rs2::error'
  what():  Frame didn't arrived within 15000
Aborted (core dumped)
```

Again, thank you for looking into this.

dorodnic, 2019-08-19:
If you suspect there is something wrong deep within librealsense, that would be an excellent opportunity to follow my advice of checking the hardware with standard Linux tools, such as yavta.

Just to set expectations straight:
1. RealSense does not officially support all possible platforms. We test on x86 and both firmware and software regularly undergo fairly significant reliability tests. 
2. This library is 100% open-source and the dev team works with the community to help enable RealSense on 3rd-party platforms. Each platform comes with its own set of quirks and features, but there are many success stories from customers running librealsense on Arm in production. 

Based on experience with this technology, I strongly recommend to rule out frames not making it out of Video4Linux subsystem, be it via yavta, v4l2 sample code or uvc logs (that I can help you enable) We do not have this specific board in our lab at the moment, so your cooperation is important. 

Thank you,

dgo-pi, 2019-08-19:
Hi Sergey,

   Thank you so much for your prompt response! We cross-built yavta, copied onto our ARM target system.

   A webcam from another vendor can be started, stopped, and restarted many times without issues on target system using yavta.

   Plugging D435 into same USB port on the target system, will work the first time the camera is plugged in. By work we mean that the D435 is emitting video data, and the IR projector is active.

   Leaving yavta, then attempting to restart D435 results in no frames being emitted, and the IR projector is off.

   Unplugging, pause, then re-inserting D435 will allow again allow frame captures and IR projector is active.

   These results seems to indicate an issue with how the D435 interacts with V4L2. Separately, there may be an error handling issue in librealsense when frame timeouts occur, but does not appear to be the core issue at hand.

   These tests were run with 2 different D435 cameras, one with firmware revision 05.09.02.00, and one with firmware revision 05.11.11.100.

   What are the suggested next steps?

dgo-pi, 2019-08-19:
(Sorry, accidently pressed wrong close & comment instead of comment)

dorodnic, 2019-08-20:
Thanks, I'm not sure what was the reason for gdb warning. Regarding the camera getting stuck after stop-start, we cannot rule out firmware issue at this moment. 
There was somewhat similar issue when closing Depth+RGB streaming abruptly (using Ctrl+C).
Some additional information that can help narrow down the problem is:
a. If the board has more than one USB3 port, does the problem repeat on all of them? 
b. If the board has USB2 ports, is the same true for when the camera is connected via USB2?
c. If the board has a PCIe slot and you have access to PCIe USB extension card, does the problem reproduce when connecting not via the built-in USB root-hub?
d. If you have access to external USB hub, does the problem reproduce when camera is connect via hub?

dgo-pi, 2019-08-20:
Hi Sergey,

   Thank you again for your fast and professional response!

_> Thanks, I'm not sure what was the reason for gdb warning. Regarding the camera getting stuck after stop-start, we cannot rule out firmware issue at this moment._

   We suspect the gdb warning is a corrupted stack, originating somewhere down in librealsense.

   We agree on the camera firmware suspicion.

_> There was somewhat similar issue when closing Depth+RGB streaming abruptly (using Ctrl+C)._

   Please explain - others are having this or a similar issue?

_> Some additional information that can help narrow down the problem is:_

_> a. If the board has more than one USB3 port, does the problem repeat on all of them?_

   Answer: We just repeated test with webcam, and with D435 (firmware version 05.11.11.100) on each of the USB ports. Same exact results.

_> b. If the board has USB2 ports, is the same true for when the camera is connected via USB2?_

   Answer: No USB2, we have all USB3 ports running at 4800 Mbps on our system.

_> c. If the board has a PCIe slot and you have access to PCIe USB extension card, does the problem reproduce when connecting not via the built-in USB root-hub?_

   Answer: No PCIe slot on our system.

_> d. If you have access to external USB hub, does the problem reproduce when camera is connect via hub?_

   Answer: Connected a commercial USB 3 hub to our system, exactly the same results with both the D435 and the webcam. This seems to rule out any hardware issue(s) on our end.

   Thanks again!

   We agree, this *does* seem to point to some kind of D435 firmware issue.

   Any further suggestions? What are the next steps?

Regards,
   David

dorodnic, 2019-08-20:
Please contact me at sergey.dorodnicov@intel.com

closed 2024-03-29 by MartyG-RealSense