realsenseai/librealsense #4656: undefined reference to udev_*

nikhilkoditala, 2019-08-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | - | 
| Operating System & Version |   Raspbian  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  Not Applicable | 
| Language                            | python| 
| Segment			| VR                       | 

### Issue Description
I am trying to install pyrealsense2 on raspberry pi 4 from source. while running the `make` command I am getting the following error. Can anyone guide me what was wrong?

`[ 59%] Linking CXX executable rs-hello-realsense
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_monitor_unref
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_enumerate_add_match_subsystem
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_enumerate_get_list_entry
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to __atomic_fetch_add_8
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_monitor_receive_device
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_device_get_sysname
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_device_get_action
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_unref
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_monitor_enable_receiving
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_monitor_filter_add_match_subsystem_devtype
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_device_unref
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_enumerate_scan_devices
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_new
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_list_entry_get_name
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_list_entry_get_next
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_device_new_from_syspath
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to __atomic_store_8
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to __atomic_load_8
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_monitor_get_fd
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_device_get_devnode
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_enumerate_unref
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_enumerate_new
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_enumerate_add_match_property
/usr/bin/ld: ../../librealsense2.so.2.25.0: undefined reference to udev_monitor_new_from_netlink
collect2: error: ld returned 1 exit status
make[2]: *** [examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/build.make:91: examples/hello-realsense/rs-hello-realsense] Error 1
make[1]: *** [CMakeFiles/Makefile2:607: examples/hello-realsense/CMakeFiles/rs-hello-realsense.dir/all] Error 2
make: *** [Makefile:130: all] Error 2`

RealSenseCustomerSupport, 2019-09-09:
----------------------------------------------

Hi @nikhilkoditala,

Please make sure that you have all the dependencies needed for successful compilation and linking:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Thank you!

RealSenseCustomerSupport, 2019-09-25:
----------------------------------------------

Hi @nikhilkoditala,

Do you need any further clarifications regarding this topic?

closed 2019-10-03 by RealSenseCustomerSupport