realsenseai/librealsense #4657: Embedding RealSense stream in Tkinter using PIL and Numpy

alifeinbinary, 2019-08-18:
I'm trying to embed the incoming colour image and depth image within a Tkinter GUI using Python. Normally, I would use the Python Image Library (PIL) to handle images but the Image.fromarray() function can't handle the RealSense data. I need to build a GUI around my RealSense feed, am I going about it the wrong way?

```
    # Streaming loop
    try:
        while True:
            # Get frameset of color and depth
            frames = pipeline.wait_for_frames()
            # frames.get_depth_frame() is a 640x480 depth image
            
            # Align the depth frame to color frame
            aligned_frames = align.process(frames)
            
            # Get aligned frames
            aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
            color_frame = aligned_frames.get_color_frame()
            
            # Validate that both frames are valid
            if not aligned_depth_frame or not color_frame:
                continue
            
            depth_image = np.asanyarray(aligned_depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())
            
            # Render images
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
            img = Image.fromarray(color_image)
            imgtk = ImageTk.PhotoImage(image=img)

            canvas = tk.Canvas(root,width=640,height=480)
            canvas.pack()
            canvas.create_image(0,0, anchor="nw", image=imgtk)

    finally:
        pipeline.stop()

```

lramati, 2019-08-29:
With the caveat that I'm not familiar with Tkinter, the code looks like it should be fine. What kind of errors are you getting?

RealSenseCustomerSupport, 2019-09-23:
----------------------------------------------

@alifeinbinary Could you please specify the details of the errors? Thanks!

RealSenseCustomerSupport, 2019-10-16:
----------------------------------------------

@alifeinbinary Sorry we didn't get reply from you for weeks. Will close it at this point. Please feel free to create new ticket if you still have issue or questions. Thanks!

closed 2019-10-16 by RealSenseCustomerSupport

samhoff20, 2019-12-10:
Why do you say that the Image.fromarray() function can't handle the Realsense data? Are you talking about only the color image? I'm experiencing an issue where it doesn't appear that my model is processing the frames, and I believe that its happening because what I'm passing isn't a PIL image but I'm not sure how to test this theory.