realsenseai/librealsense #4659: Where is the Intel.Realsense.DynamicCalibrator source file

McJun1, 2019-08-19:
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435 | 
| Operating System & Version | ubuntu16.04 | 

I want to use the robot arm and this software(Intel.Realsense.DynamicCalibrator) to automate the calibration of the depth camera.I want to modify source file to connect my robot arm, but I can't find where the Intel.Realsense.DynamicCalibrator source file is. Can u tell me ? Thanks.

dorodnic, 2019-08-19:
Hi @McJun1 
Dynamic Calibration is not open-source. 
We are working on a cross-platform alternative in upcoming firmware release (#4658) and also on cross-compiling DC to Arm (can't give any specific schedule on this one, it really depends on customer feedback). Could you please share what type of platform are you using?

RealSenseCustomerSupport, 2019-09-05:
----------------------------------------------

@McJun1, please let us know if there is any update or needing further assistance.
Thank you.