realsenseai/librealsense #4660: Inconsistent playback behavior with pyrealsense2

tanikapod, 2019-08-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.23.0  | 
| Language                            |  python 2.7 (because python 3 apparently doesn't work for pyrealsense2 on Windows) | 

I am recording both the depth stream and the IMU acceleration stream into a .bag file, and then playing back the recording to process the frames. Because I need to process each frame, I am using `playback.set_real_time(False)`, but since that didn't seem to fix the issue, I  am also using `playback.pause()` and `playback.resume()` on either end of the frame processing.

The inconsistency is that I seem to get different frames (sometimes a different total number of frames) as well as different timestamps for those frames each time I run the script. Below is the general outline of how I'm implementing my playback feature, minus some irrelevant stuff taken out - am I doing something wrong? (Also related - is there a better way to get the acceleration frame than looping through all the frames and checking `is_motion_frame`? I can't find a `get_motion_frame` or anything similar to `get_depth_frame`, which would be nice)

```
config = rs.config()
rs.config.enable_device_from_file(config, settings.data_directory, repeat_playback = False)
self.pipeline = rs.pipeline()

config.enable_stream(rs.stream.depth, settings.resolution_width, settings.resolution_height, settings.depth_format, settings.frames_per_sec)
config.enable_stream(rs.stream.accel, settings.accel_format, settings.accel_fps)

profile = self.pipeline.start(config)
self.playback = profile.get_device().as_playback()
self.playback.set_real_time(False)

try:
    while True:
        try: frames = self.pipeline.wait_for_frames()
        except: break   # assume no more frames in bag file

        self.playback.pause()

        for frame in frames:
            if frame.is_motion_frame():
                accel_data = frame.as_motion_frame().get_motion_data()
                x, y, z = accel_data.x, accel_data.y, accel_data.z
                break
        self.frames.append(frames.get_depth_frame())

        self.playback.resume()

finally:
    self.pipeline.stop()
```

lramati, 2019-08-21:
The code looks fine.
As for `get_motion_frame`, while no such convenience function exists, I suggest you take a look at [`rs.composite_frame.first_or_default`](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.composite_frame.html#pyrealsense2.composite_frame.first_or_default) which is used to [implement](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_frame.hpp#L945) the other convenience functions

tanikapod, 2019-08-22:
Thanks, I'll look into that!

I guess what I'm asking is how is it possible that each time I run my code, there are a different total number of frames, each with slightly inconsistent timestamps?

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@tanikapod Did you try https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector to see if any incosistency? Looking forward to your update. Thanks!

tanikapod, 2019-11-25:
Sorry, this was actually for a summer project that I'm not working on anymore. I'll close the issue.

closed 2019-11-25 by tanikapod