realsenseai/librealsense #4661: how to make hole filling of partial region?

jjh930910, 2019-08-19:
* 
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {windows10  | 
| Kernel Version (Linux Only)    |  ()                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {  }                          | 
| Language                            |  {/python/ }                          | 
| Segment			|  {pupil tracking }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

i want to get depth map of partial region with hole filling because i save wasting time in hole filling preprocessing

so i need to get depth map of only face region with hole filling  ( i have already made face detection algorithm )

and i want to make hole filling filter in preprocessing   

so i want to hear your advice

dorodnic, 2019-08-19:
Hi @jjh930910 
At the moment there is no such capability in the SDK, although it makes a lot of sense and we are going to add such API in the future. But for now your options are to apply post processing on the full frame or crop and implement something similar using OpenCV on top of librealsense.

jjh930910, 2019-08-19:
because i am beginner of computer vision , i have one more question  

i tried to crop images and hole filling with opencv but opencv function is binary hole filling  

is it similar function to Intel realsense library hole filling filter?

g2-bernotas, 2019-08-19:
I have used [opencv inpaint function](https://docs.opencv.org/3.3.1/df/d3d/tutorial_py_inpainting.html) with good results. First, you will have to identify where the holes. This can be done in a series of steps:

1. Convert your depth image into a binary image. 
2. Apply an opencv fill operation (e.g. [floodFill](https://docs.opencv.org/2.4/modules/imgproc/doc/miscellaneous_transformations.html?highlight=floodfill)). 
3. Subtract the original binary image from the filled image to receive a binary image indicating the locations of holes. 
4. Apply [dilation morphological operation](https://docs.opencv.org/2.4/doc/tutorials/imgproc/erosion_dilatation/erosion_dilatation.html). This showed to work better than without it. 
5. Finally, apply [inpaint](https://docs.opencv.org/3.3.1/df/d3d/tutorial_py_inpainting.html) function to your grayscale image also providing the location of holes. I chose radius of 10 and INPAINT_TELEA algorithm. Here is the Python line to call this function:

``` Python
destination = cv2.inpaint(your_img, binary_holes, 10, cv2.INPAINT_TELEA)
```

jjh930910, 2019-08-19:
oh thank you ,g2-bernotas  i will try this function

RealSenseCustomerSupport, 2020-02-28:
----------------------------------------------

Hi @ jjh930910 Do you have any update?
If we don't hear from you in 7 days, we will close this issue.

closed 2020-07-02 by RealSenseSupport