realsenseai/librealsense #4664: param-amplitude-factor key is not supported by the connected device at rs2_load_json

NXTwoThou, 2019-08-19:
Camera Model: D415
Firmware Version: 05.11.11.100
Operating System & Version: Windows 10 18963
SDK Version: 2.25
Language: C#

I was having an issue where I couldn't use any presets with my camera.  A new Windows 10 build came out on Friday and it now allows me to select a preset in the RealSense Viewer.  The new issue is I can't load up a preset in my own code.

I can load my json file in the realsense viewer, turn on stereo module and rgb camera and everything looks as expected.  I can then save the preset over the existing file.  When I try to 

ctx = new Context();
            var devices = ctx.QueryDevices();
            if (devices.Count > 0)
            {
                Device dev = devices[0];
                AdvancedDevice ad = AdvancedDevice.FromDevice(dev);
                ad.JsonConfiguration = File.ReadAllText("GanderD415-4.json");
            }

I get an error of "param-amplitude-factor key is not supported by the connected device at rs2_load_json".  If I edit the json and get rid of the param-amplitude-factor key, I get a "hr returned: HResult 0x8007001f: "A device attached to the system is not functioning at rs2_load_json".

If I return to RealSense Viewer, load up and resave out the json, it puts back the param-amplitude-factor key.

If I go to https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets and attempt to load ShortRangePreset.json I get the same  I get a "hr returned: HResult 0x8007001f: "A device attached to the system is not functioning at rs2_load_json".  If I use ShortRangePreset.json in RealSense Viewer, it loads fine, if I overwrite the json from Viewer, I, of course, get the param-amplitude-factor error.

If I don't set JsonConfiguration, it loads fine(but of course is unusable as I don't have my settings)

Since I'm trying to do extreme close up scans, I'd really like Subpixel Linearity(depth linearity, disparity modulation, amplitude-factor, whatever you want to call it, it's painful to try and find something when it's named so many different ways depending on where you look).  At this point I'd like to at least be able to load any preset file.

Here's my json file(even though it shouldn't matter since it doesn't work with the files downloaded from https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets )

{
    "aux-param-autoexposure-setpoint": "200",
    "aux-param-colorcorrection1": "-0.878906",
    "aux-param-colorcorrection10": "-0.342773",
    "aux-param-colorcorrection11": "-0.342773",
    "aux-param-colorcorrection12": "0.0537109",
    "aux-param-colorcorrection2": "1.49707",
    "aux-param-colorcorrection3": "1.49707",
    "aux-param-colorcorrection4": "-1.71777",
    "aux-param-colorcorrection5": "-0.166992",
    "aux-param-colorcorrection6": "-0.0888672",
    "aux-param-colorcorrection7": "-0.0888672",
    "aux-param-colorcorrection8": "0.344727",
    "aux-param-colorcorrection9": "0.633789",
    "aux-param-depthclampmax": "3000",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "200",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "33000",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "156",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-depth-white-balance-auto": "False",
    "controls-laserpower": "360",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0.08",
    "param-autoexposure-setpoint": "200",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "3000",
    "param-depthclampmin": "0",
    "param-depthunits": "100",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "0",
    "param-disparitymode": "0",
    "param-disparityshift": "200",
    "param-lambdaad": "800",
    "param-lambdacensus": "26",
    "param-leftrightthreshold": "10",
    "param-maxscorethreshb": "2048",
    "param-medianthreshold": "0",
    "param-minscorethresha": "10",
    "param-neighborthresh": "0",
    "param-raumine": "0",
    "param-rauminn": "0",
    "param-rauminnssum": "0",
    "param-raumins": "0",
    "param-rauminw": "0",
    "param-rauminwesum": "0",
    "param-regioncolorthresholdb": "0.000978474",
    "param-regioncolorthresholdg": "0.000978474",
    "param-regioncolorthresholdr": "0.000978474",
    "param-regionshrinku": "0",
    "param-regionshrinkv": "0",
    "param-robbinsmonrodecrement": "0",
    "param-robbinsmonroincrement": "0",
    "param-rsmdiffthreshold": "4",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0",
    "param-scanlineedgetaub": "72",
    "param-scanlineedgetaug": "72",
    "param-scanlineedgetaur": "72",
    "param-scanlinep1": "60",
    "param-scanlinep1onediscon": "105",
    "param-scanlinep1twodiscon": "70",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "190",
    "param-scanlinep2twodiscon": "130",
    "param-secondpeakdelta": "0",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-zunits": "100",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}

ev-mp, 2019-08-19:
@NXTwoThou hello,
I cannot immediately comment on the Insider's preview build that you're using, but I've tried the preset that you shared on a stable Win10/ Redstone3 (build 16299) using D415 with identical v5.11.11.100 FW: The preset could be loaded, executed and exported properly both in USB2 and USB3 modes, including the `A-factor` flag.

Can you check whether running the code on Win10 Redstone 3/4 builds would mitigates this?

NXTwoThou, 2019-08-19:
Testing with the realsense viewer unfortunately doesn't help as the realsense viewer(as of Friday) works again on this machine.  Using IntelRealsense.dll v2.25.0.0 in a C# app crashes on JsonConfiguration=

ev-mp, 2019-08-19:
@NXTwoThou, please clarify -  are you able to load this preset correctly using the `realsense-viewer` but not the C# wrapper on that specific PC ?
Or is it the case that the preset cannot be loaded with neither C# nor `realsense-viewer` ?

NXTwoThou, 2019-08-19:
realsense-viewer works fine

C# wrapper does not

Saving preset from realsense-viewer generates "param-amplitude-factor key is not supported by the connected device at rs2_load_json" error in C# wrapper

Removing param-amplitude-factor key from json preset file allows C# wrapper to get a little further by giving a "hr returned: HResult 0x8007001f: "A device attached to the system is not functioning at rs2_load_json"

NXTwoThou, 2019-08-21:
Unfortunately same issue with v2.26.0.  

1) RealSense Viewer works fine without loading a preset
2) RealSense Viewer works fine loading my custom preset file.  
3) Using the C# wrapper gives an error about an invalid parameter when trying to set JsonConfiguration.  
4) If I remove the parameter from the json file, setting JsonConfiguration says "a device attached to the system is not functioning"
5) If I remove trying to set JsonConfiguration, it runs fine(only without the presets I require)

Here's some images since my original description might not have been understood clearly...

1) RealSense Viewer with "Default"(no) preset:
[image]

2) RealSense Viewer with "GanderD415-4.json" preset:(contains param-amplitude-factor)
[image]

3) My code first error
[image]
Detailed error view
```
System.ArgumentException
  HResult=0x80070057
  Message=param-amplitude-factor key is not supported by the connected device!
  Source=Intel.RealSense
  StackTrace:
   at Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged(IntPtr pNativeData) in C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\wrappers\csharp\Intel.RealSense\Helpers\ErrorMarshaler.cs:line 72
   at System.StubHelpers.MngdRefCustomMarshaler.ConvertContentsToManaged(IntPtr pMarshalState, Object& pManagedHome, IntPtr pNativeHome)
   at Intel.RealSense.NativeMethods.rs2_load_json(IntPtr dev, String json_content, UInt32 content_size, Object& error)
   at Intel.RealSense.AdvancedDevice.set_JsonConfiguration(String value) in C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\wrappers\csharp\Intel.RealSense\Devices\AdvancedDevice.cs:line 84
Inner Exception 1:
ExternalException: rs2_load_json(dev:000002851D9D99F0, json_content:000002851DABAEC0, content_size:3615)
```

4) My code after removing param-amplitude-factor from json file
[image]
Detailed error view
```
System.Exception
  HResult=0x80131500
  Message=hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
  Source=Intel.RealSense
  StackTrace:
   at Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged(IntPtr pNativeData) in C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\wrappers\csharp\Intel.RealSense\Helpers\ErrorMarshaler.cs:line 78
   at System.StubHelpers.MngdRefCustomMarshaler.ConvertContentsToManaged(IntPtr pMarshalState, Object& pManagedHome, IntPtr pNativeHome)
   at Intel.RealSense.NativeMethods.rs2_load_json(IntPtr dev, String json_content, UInt32 content_size, Object& error)
   at Intel.RealSense.AdvancedDevice.set_JsonConfiguration(String value) in C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\wrappers\csharp\Intel.RealSense\Devices\AdvancedDevice.cs:line 84
   at Gander.RealSense.RealSense_Load(Object sender, EventArgs e) in Inner Exception 1:
ExternalException: rs2_load_json(dev:000001D37DD269E0, json_content:000001D37DDCE5B0, content_size:3576)
```

5) My code after commending out JsonConfiguration line
[image]

NXTwoThou, 2019-08-29:
Same error with v2.27.0

[image]

NXTwoThou, 2019-08-30:
Same issue with build 18970

NXTwoThou, 2019-09-05:
Same issue with 2.28.0

NXTwoThou, 2019-09-05:
Hit a pausing point in my primary project so I could take a real look at this.  I went ahead and downloaded the latest source so I could try and debug what was going on.  I discovered the error was 100% me being a complete idiot.

When I first created my project I had added a reference to C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64\Intel.Realsense.dll to my project.  When I compile, it of course copies the latest Intel.Realsense.dll to my folder.  When I first tried to run my project, I would have gotten an error that realsense2.dll isn't found.   I then copied realsense2.dll from that folder to my output folder.  Looks like I started this back in April as that was the file date of realsense2.dll

Visual Studio will copy the most recent Intel.Realsense.dll on every compile to my output folder.  Unfortunately it doesn't check it's dependencies.  I've gone ahead and added it to my solution as a build action of "Content" and Copy to Output Directory of "Copy if newer".   I feel incredibly stupid.  I suggest adding this to the example projects.

closed 2019-09-05 by NXTwoThou