realsenseai/librealsense #4665: Please share the steps for kernel version. 4.9 for NVIDIA Drive PX2

RatheeshRK, 2019-08-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.9 (cannot be changed as per NVIDIA feedback                                     | 
| Platform                                 | NVIDIA Drive PX2  |

### Issue Description
<Intel realsense D435 camera, instalation on Drive PX2, Tegra B | Kernel update error.>

Feedback from NVIDIA: https://devtalk.nvidia.com/default/topic/1058261/drive-linux/intel-realsense-d435-camera-instalation-on-drive-px2-tegra-b-kernel-update-error/post/5368554/#5368554

Need help to install Realsense D435 in NVIDIA Drive PX2 with Kernel 4.9.

dorodnic, 2019-08-19:
Hi @RatheeshRK 
Our current strategy for NVidia line of devices is using the [libuvc backend](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md). This method  does not require any kernel patching. I don't have experience with the Drive PX2 device, but if it is anything like the Jetson family, there is a decent chance it may even work out of the box with [regular Debian installation instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). 

We are working on expanding support for this method (upcoming release is going to add the D435i as well), but there are known issues with RealSense on some NVidia devices, particularly related to USB connectivity and stability. Please share your experience.

dorodnic, 2019-08-27:
Hi @RatheeshRK 
Any luck with this? I'm curious to know how it went

closed 2019-09-16 by dorodnic