realsenseai/librealsense #4667: librealsense/ros-realsense error at run-time : undefined symbol: libusb_get_parent

harshitsankhla, 2019-08-19:
**OS : Ubuntu 16.04.6
ROS : Kinetic
Camera - D435**

For the past few days I have tried to build realsense library (source and debian packages) and the ros wrapper on 2 separate computers (Intel NUCs, the irony !) I am able to build both the packages but on launch (roslaunch realsense2_camera rs_rgbd.launch) I keep getting the error (most of the times, and most recently stuck here after a fresh install) - 

_/opt/ros/kinetic/lib/nodelet/nodelet: symbol lookup error: /usr/lib/x86_64-linux-gnu/librealsense2.so.2: undefined symbol: libusb_get_parent_

**Most Recent Install sequence -**

version="2.18.1-0~realsense0.568"
sudo apt-get -y install librealsense2-udev-rules=${version}
sudo apt-get -y install librealsense2-dkms=1.3.4-0ubuntu1
sudo apt-get -y install librealsense2=${version}
sudo apt-get -y install librealsense2-utils=${version}
sudo apt-get -y install librealsense2-dev=${version}
sudo apt-get -y install librealsense2-dbg=${version}

mkdir -p /rsws/src
cd rsws/src
catkin_init_workspace
git clone https://github.com/intel-ros/realsense.git
cd ..
catkin_make clean
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
echo "source ~/rsws/devel/setup.bash" >> ~/.bashrc
source ~/rsws/devel/setup.bash

**Tried Workarounds -** 

1. Format the computer and build before installing any other software or package besides ROS
2. Separately install libusb-1.0.0-dev from source and apt
3. Build from source a previous version of librealsense
4. Saw a thread mentioning a conflict with Bluefox2 camera, but that should've been resolved    when I formatted the computer and tried again.

Would appreciate any kind of help to work this out.

RealSenseCustomerSupport, 2019-09-17:
----------------------------------------------

@harshitsankhla Sorry for late reply. Did you still get the issue? Any luck with the latest SDK and ROS version? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2019-09-29:
----------------------------------------------

@harshitsankhla Do you still need support on this? Could you please update? Thanks!

harshitsankhla, 2019-10-09:
Hi

Apologies for the late reply.

I have figured out what the issue was. Basically real sense(the library and applications based on it) use dynamically linked library - _libusb-dev_ 

Now some applications can install their own version of this library (in my case - the drive for mvImpact Bluefox camera) and override the path to the library for all other applications by modifying a path variable or adding a file in /etc/conf

So you have to double-check that the all the dynamic link library locations are exactly the ones which are set during the library installation and no other application has modified them (ldd command in linux can be handy)

closed 2019-10-09 by harshitsankhla