realsenseai/librealsense #4669: Failing to build patched uvcvideo for 4.15.0-58-generic

phil-marble, 2019-08-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |   4.15.0-58-generic, x86_64                                | 
| Platform                                 | PC  |
| SDK Version                            |       2.15.0                      | 
| Language                            |  N/A                | 
| Segment			|  Robot                       | 

### Issue Description

librealsense-dkms kernel module fails to build:

```
Building initial module for 4.15.0-58-generic
Error! Bad return status for module build on kernel: 4.15.0-58-generic (x86_64)
Consult /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log for more information.

```
$ cat /var/lib/dkms/librealsense2-dkms/1.3.5/build/make.log
DKMS make.log for librealsense2-dkms-1.3.5 for kernel 4.15.0-58-generic (x86_64)
Mon Aug 19 20:54:26 UTC 2019
make: Entering directory '/usr/src/linux-headers-4.15.0-58-generic'
Makefile:986: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_driver.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_queue.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_v4l2.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_video.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_ctrl.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_status.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_isight.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_debugfs.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvc_entity.o
  LD [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvcvideo.o
  MODPOST 1 modules
  CC      /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvcvideo.mod.o
  LD [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/usb/uvc/uvcvideo.ko
make: Leaving directory '/usr/src/linux-headers-4.15.0-58-generic'
make: Entering directory '/usr/src/linux-headers-4.15.0-58-generic'
Makefile:986: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-dev.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-ioctl.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-device.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-fh.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-event.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-ctrls.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-subdev.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-clk.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-async.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-compat-ioctl32.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/vb2-trace.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-trace.o
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/v4l2-mc.o
  LD [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/videodev.o
  MODPOST 1 modules
  CC      /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/videodev.mod.o
  LD [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/media/v4l2-core/videodev.ko
make: Leaving directory '/usr/src/linux-headers-4.15.0-58-generic'
make: Entering directory '/usr/src/linux-headers-4.15.0-58-generic'
Makefile:986: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/iio/accel/hid-sensor-accel-3d.o
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/iio/accel/hid-sensor-accel-3d.c: In function ‘accel_3d_read_raw’:
/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/iio/accel/hid-sensor-accel-3d.c:163:11: error: too few arguments to function ‘sensor_hub_input_attr_get_raw_value’
    *val = sensor_hub_input_attr_get_raw_value(
           ^
In file included from /var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/iio/accel/hid-sensor-accel-3d.c:26:0:
./include/linux/hid-sensor-hub.h:191:5: note: declared here
 int sensor_hub_input_attr_get_raw_value(struct hid_sensor_hub_device *hsdev,
     ^
scripts/Makefile.build:337: recipe for target '/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/iio/accel/hid-sensor-accel-3d.o' failed
make[1]: *** [/var/lib/dkms/librealsense2-dkms/1.3.5/build/4.15.0/drivers/iio/accel/hid-sensor-accel-3d.o] Error 1
Makefile:1738: recipe for target 'hid-sensor-accel-3d.ko' failed
make: *** [hid-sensor-accel-3d.ko] Error 2
make: Leaving directory '/usr/src/linux-headers-4.15.0-58-generic'
```

ev-mp, 2019-08-19:
@phil-marble hello,
The latest update in the LTS 4.15 branch (4.15.0-58) introduced a breaking change in one of the kernel call signatures:
>too few arguments to function ‘sensor_hub_input_attr_get_raw_value’

 This currently breaks the DKMS package for that specific version.
We're working on a fix, in the meanwhile you can continue using the previous 4.15.0-55 kernel from that branch. Alternatively, you may install the attached WA DKMS  that is designed to run on 4.15.0-58 but fail previous versions (e.g. 4.15.0-55)
[librealsense2-dkms_1.3.5-0ubuntu1_all.deb.txt](https://github.com/IntelRealSense/librealsense/files/3517746/librealsense2-dkms_1.3.5-0ubuntu1_all.deb.txt)
```bash
$sudo dpkg -i librealsense2-dkms_1.3.5-0ubuntu1_all.deb
```
Remove the ".txt" suffix before installation

ev-mp, 2019-11-14:
The fix for the issues is available for download via PPA.
To upgrade from previous version use `sudo apt update && sudo apt upgrade`.,
For full reinstall use:
`sudo apt purge librealsense2-dkms && sudo apt update && sudo apt install librealsense2-dkms`

ev-mp, 2019-12-01:
@phil-marble, do you still need assistance with dkms for 4.15 ? 
Please update

phil-marble, 2019-12-02:
@ev-mp I have been unable to test yet, but thanks for the update!

ev-mp, 2020-01-26:
Assume resolved.

closed 2020-01-26 by ev-mp