realsenseai/librealsense #4670: Generating Pointclouds from Numpy arrays

LossOfHumanIdentity, 2019-08-20:
Hi, 

I would like to obtain pointcloud not from a frame, but from a conventional numpyarray. 

I untitled the question with the exact same title of an old question, as I would like to understand if this has been implemented/solved or not (I think this is a really important feature).

I am basically trying to render the 3D Pointcloud not of the whole scene, but only of a certain region of interest ROI, that is why I need to be able to obtain a pointcloud starting from the resulting numpyarray image. 

I look forward to hear from you. 

Best Regards

MartyG-RealSense, 2019-08-20:
Is the link below of any help to you, please?

https://github.com/IntelRealSense/librealsense/issues/1466

LossOfHumanIdentity, 2019-08-20:
Hi, thank you a lot for your message. Actually my question addresses computing a point cloud not from a realsense frame, but from a standard opencv disparity map image. Would you have any hint about that? Thanks in advance

MartyG-RealSense, 2019-08-20:
I recall a case in the link below where someone was trying to generate a disparity map in OpenCV and converting the depth points into 3D points.

https://forums.intel.com/s/question/0D50P0000490TLaSAM/issues-with-custom-disparity-generation?language=en_US

RealSenseCustomerSupport, 2019-09-24:
----------------------------------------------

Hi @ LossOfHumanIdentity,

Has the link provided by MartyG-RealSense helped to find an answer to your question?

RealSenseCustomerSupport, 2019-10-03:
----------------------------------------------

@LossOfHumanIdentity

Do you need any further clarifications regarding this topic?

closed 2019-10-22 by RealSenseCustomerSupport